solettaproject/soletta #1985: Add support for IIO event

lblim, 2016-05-06:
#### Task Description
Please add IIO event support into IIO library as some sensors is supported via IIO event.

lblim, 2016-05-10:
This is needed for SX9500 proximity sensor support.

barbieri, 2016-05-10:
@yongli3 can you handle this as well (or delegate to a team member)? @edersondisouza can handle the review (he is currently allocated to a different task, would take some time to get to it)

yongli3, 2016-05-11:
@barbieri No problem! Let us works on this one and #1998, and refactor the nodes.c as well

lblim, 2016-06-24:
May I know whether this will be included in v1 or v2?

barbieri, 2016-06-24:
v2, since v1 features are frozen so we can get a stable release. But it should be easy to do, so should get in the repo soon.

Also, v2 is not expected to take lots of months, likely 2-3 months to get the next stable release. Depends on the bandwidth and number of changes :-)

lblim, 2016-07-13:
can this be tagged for v2 milestone?

yongli3, 2016-07-14:
Reference code http://lxr.free-electrons.com/source/tools/iio/iio_event_monitor.c

yongli3, 2016-07-14:
After discussed with @laykuanloon, this is low priority, we will focus on #2205 first