solettaproject/soletta #1998: IIO node type: add support for multiple channels

edersondisouza, 2016-05-09:
#### IIO node type: add support for multiple channels

Currently, IIO node types support only one logic channel. Some devices, however, have more than one channel, whose readings are independent. They usually have channels `in_readingX`, like `in_voltage0` or `in_proximity2`.
To address that, IIO node types would need to have a new option, `channel`, default -1. If -1, it means that are no specifier, so channel name would be `in_proximity`. If > 0, then this should be used as channel specifier.

### Implementation notes

Some thoughts on implementation, of course different ideas are welcome.
Even though the channels are independent, they do belong to the same device. So, it's necessary that different nodes share devices. This can be done by creating a list of open devices on nodes.c.
After opening a device, it shall be sent to this list. I believe that this list should store device id, sol_iio_device handle and device/channel config. Last one is to prevent that some channel is opened with different device config - so it's there to compare.
So, when creating a node, it should first call `sol_iio_address_device`, and check if there's an sol_iio_device open with same id. If so, compare configs and if OK, add new channel. If not, fail on mismatch configuration.
This is also an opportunity to make iio/nodes.c code share more common code and avoid it's current state of duplicating code.

edersondisouza, 2016-05-09:
@yongli3 See this issue, please.

lblim, 2016-05-10:
This will allow multiple access, will the implementation only allow multiple access to the same IIO device from a single application or it will allow multiple access from multiple applications? Where is the access controller here.

yongli3, 2016-05-10:
@edersondisouza I am thinking we need to add another option: count. Default = 1, it means we only output one channel. > 1 means output multi-channel data. Using the ADC sensor on Galileo as an example. Below is the sysfs entities:
ADC sensor on Galileo boards:
root@intel-quark:/sys/bus/iio/devices/iio:device0# ls -l
drwxr-xr-x    2 root     root             0 May  2 06:44 buffer
-r--r--r--    1 root     root          4096 May  2 06:44 dev
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage0_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage0_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage1_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage1_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage2_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage2_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage3_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage3_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage4_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage4_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage5_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage5_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage6_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage6_scale
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage7_raw
-rw-r--r--    1 root     root          4096 May  2 06:44 in_voltage7_scale
-r--r--r--    1 root     root          4096 May  1 10:41 name
drwxr-xr-x    2 root     root             0 May  2 06:44 power
drwxr-xr-x    2 root     root             0 May  2 06:44 scan_elements
lrwxrwxrwx    1 root     root             0 May  2 06:44 subsystem -> ../../../../../../../../bus/iio
drwxr-xr-x    2 root     root             0 May  2 06:44 trigger
-rw-r--r--    1 root     root          4096 Apr 28 05:51 uevent

Without channel and count options, only in_voltage0_raw is captured(The current implementation). 
If channel=0, count=1, it will capture in_voltage0_raw(The same as default). 
If channel=0, count=2, it will capture in_voltage0_raw and in_voltage1_raw, and output two values
If channel=1, count=1, it will capture in_voltage1_raw only
If channel=3, count=3, it will capture in_voltage3_raw, in_voltage4_raw and in_voltage5_raw

For below color sensor, by default all 3 data are captured. If channel=1, count=1, it will only output in_intensity_green

in_intensity_red
in_intensity_green
in_intensity_blue

The same as below magnet sensor, by default all x/y/z data are captured, but if If channel=1, count=2, it will only output in_magn_y_raw and in_magn_z_raw

in_magn_x_raw
in_magn_y_raw
in_magn_z_raw

Any thoughts/comments?

edersondisouza, 2016-05-10:
> This will allow multiple access, will the implementation only allow
> multiple access to the same IIO device from a single application or it
> will allow multiple access from multiple applications? Where is the
> access controller here.

I was thinking about multiple access from a single application. Having 
different nodes for different data coming from the same device.

edersondisouza, 2016-05-10:
> @edersondisouza <https://github.com/edersondisouza> I am thinking we
> need to add another option: count. Default = 1, it means we only output
> one channel. > 1 means output multi-channel data. Using [...]

I can see value in this idea, it would help to save some memory for 
devices that expose 'indexed' channels like those voltageX or proximityY 
ones. But not sure if this will end up bringing more confusion to users. 
If someone needs only 'voltage0' and 'voltage7' it would be no better 
than having two nodes, anyway.

> For below color sensor, by default all 3 data are captured. If
> channel=1, count=1, it will only output in_intensity_green
>
> in_intensity_red
> in_intensity_green
> in_intensity_blue
>
> The same as below [...]

I'm not sure I see value on these ones, though. Red, green and blue and 
x, y and z are actually part of the *same* information, it doesn't make 
sense separate them. Actually, node types output only one packet 
containing all information: rgb packet for first case and 
direction-vector packet for last.

yongli3, 2016-05-10:
@edersondisouza I got your point, some information cannot be separated. Let us add the channel option only. I will use the ADC sensor for a test

edersondisouza, 2016-05-12:
One more thing to notice: AFAIR, one can only enable channel scanning when buffer is disabled - talking about sysfs IIO control. If that's so, one could not add a channel after device was opened, making device sharing impossible.
To overcome this, part of channel setup would have to be delayed to happen when buffer is started, for instance, `enable_channel_scan`, `read_channel_scan_info` and `read_channel_index_in_buffer`.
`sol_iio_device_start_buffer` would them change to first disable buffer, setup channels and them enable buffer as is today.
Note that this changes don't affect API and thus how nodes setup: these changes should be internal to `sol-iio.c`.

lblim, 2016-06-24:
May I know whether this support will be in v1?

lblim, 2016-07-13:
can this be included for v2 milestones?

barbieri, 2016-07-13:
sure!

peijiajames, 2016-10-27:
@edersondisouza I see the fix #2318 has been merged. Do you think it is okay to close this task now?

peijiajames, 2016-11-01:
@edersondisouza @ceolin ping.

ceolin, 2016-11-04:
since the fix was already merged, i'm closing this issue.

closed 2016-11-04 by ceolin