solettaproject/soletta #2043: Soletta library path is default to /usr/lib even for 64-bit build

lblim, 2016-05-16:
#### Issue Description

Soletta default the library directory to /usr/lib, but for our BSP the image is needed in lib64 if not image building will fail as shown in rootfs.jpg. Tweak has been done to install the libsoletta into /usr/lib64 path, but with this tweak will hit into runtime problem as the sol-fbp-runner is looking for the library under /usr/lib. Temporary solution is we create the soft link so that teh libraries appeared at both /usr/lib and /usr/lib64.

 
**soletta_git.bb**
141 do_install() {
142    oe_runmake DESTDIR=${WORKDIR}/image install CFLAGS="--sysroot=${STAGING_D    IR_TARGET}" TARGETCC="${CC}" TARGETAR="${AR}"
143    unlink ${WORKDIR}/image/usr/lib/libsoletta.so
144    mv ${WORKDIR}/image/usr/lib/libsoletta.so.0.0.1 ${WORKDIR}/image/usr/lib/    libsoletta.so
145    ln -sf libsoletta.so ${WORKDIR}/image/usr/lib/libsoletta.so.0.0.1
146    COMMIT_ID=`git --git-dir=${WORKDIR}/git/.git rev-parse --verify HEAD`
147    echo "Soletta: $COMMIT_ID" > ${D}/usr/lib/soletta/soletta-image-hash
148 
149    # Remove nan module as it is not needed.
150    rm -rf ${WORKDIR}/image/usr/lib/node_modules/soletta/node_modules/nan
151 }
[image]

[leebooil@leebooil-dev soletta]$ grep -nr MODULESDIR *
config.log:47:FLOWMETATYPEMODULESDIR = /usr/lib/soletta/modules//flow-metatype/
config.log:48:FLOWMETATYPEMODULESDIR_RELATIVE = ${modulesdir}//flow-metatype/
config.log:49:FLOWMODULESDIR = /usr/lib/soletta/modules//flow/
config.log:50:FLOWMODULESDIR_RELATIVE = ${modulesdir}//flow/
config.log:79:LINUXMICROMODULESDIR = /usr/lib/soletta/modules//linux-micro/
config.log:80:LINUXMICROMODULESDIR_RELATIVE = ${modulesdir}//linux-micro/
config.log:88:MODULESDIR = /usr/lib/soletta/modules/
config.log:89:MODULESDIR_RELATIVE = ${libdir}/soletta/modules/
config.log:138:UPDATEMODULESDIR = /usr/lib/soletta/modules//update/
config.log:139:UPDATEMODULESDIR_RELATIVE = ${modulesdir}//update/
pc/soletta.pc.in:5:modulesdir={{@MODULESDIR_RELATIVE@}}
src/shared/sol-modules.c:140:    return snprintf(buf, len, "%s%s%s/%s.so", rootdir, MODULESDIR, nspace, modname);
src/samples/flow/c-api/find-type.c:180:#ifndef MODULESDIR
src/samples/flow/c-api/find-type.c:181:#define MODULESDIR "/usr/lib/soletta/modules"
src/samples/flow/c-api/find-type.c:183:#ifndef FLOWMODULESDIR
src/samples/flow/c-api/find-type.c:184:#define FLOWMODULESDIR MODULESDIR "/flow"
src/samples/flow/c-api/find-type.c:187:    r = snprintf(path, sizeof(path), "%s" FLOWMODULESDIR "/%s.so",
src/samples/flow/c-api/find-type.c:190:        fprintf(stderr, "ERROR: path is too long %s" FLOWMODULESDIR "/%s.so\n",
src/lib/common/sol-platform-impl-linux-micro.c:171:        LINUXMICROMODULESDIR, name);
src/lib/common/sol-platform-impl-linux-micro.c:343:        LINUXMICROMODULESDIR "/initial-services",
src/lib/common/sol-update.c:128:            UPDATEMODULESDIR, "SOL_UPDATE", check_module);
src/lib/common/sol-update.c:133:        name_to_free = get_first_module_on_dir(UPDATEMODULESDIR);
src/lib/flow/sol-flow-parser-dynamic.c:59:            FLOWMETATYPEMODULESDIR, "SOL_FLOW_METATYPE", check_metatype);
src/lib/flow/sol-flow-resolver-conffile.c:181:        install_rootdir, FLOWMODULESDIR, name);
tools/build/Makefile.vars:98:export MODULESDIR := $(LIBDIR)$(PKGNAME)/modules/
tools/build/Makefile.vars:105:export FLOWMODULESDIR := $(MODULESDIR)/flow/
tools/build/Makefile.vars:106:export PINMUXDIR := $(MODULESDIR)/pin-mux/
tools/build/Makefile.vars:107:export LINUXMICROMODULESDIR := $(MODULESDIR)/linux-micro/
tools/build/Makefile.vars:108:export FLOWMETATYPEMODULESDIR := $(MODULESDIR)/flow-metatype/
tools/build/Makefile.vars:109:export UPDATEMODULESDIR := $(MODULESDIR)/update/
tools/build/Makefile.vars:118:export MODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(LIBDIR)),$${libdir},$(MODULESDIR))
tools/build/Makefile.vars:124:export FLOWMODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(MODULESDIR)),$${modulesdir},$(FLOWMODULESDIR))
tools/build/Makefile.vars:126:export LINUXMICROMODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(MODULESDIR)),$${modulesdir},$(LINUXMICROMODULESDIR))
tools/build/Makefile.vars:127:export FLOWMETATYPEMODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(MODULESDIR)),$${modulesdir},$(FLOWMETATYPEMODULESDIR))
tools/build/Makefile.vars:128:export UPDATEMODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(MODULESDIR)),$${modulesdir},$(UPDATEMODULESDIR))
tools/build/Makefile.vars:138:build_modulesdir := $(build_sysroot)$(MODULESDIR)
tools/build/Makefile.vars:304:	-DMODULESDIR="\"$(MODULESDIR)\"" \
tools/build/Makefile.vars:306:	-DFLOWMODULESDIR="\"$(FLOWMODULESDIR)\"" \
tools/build/Makefile.vars:308:	-DLINUXMICROMODULESDIR="\"$(LINUXMICROMODULESDIR)\"" \
tools/build/Makefile.vars:309:	-DFLOWMETATYPEMODULESDIR="\"$(FLOWMETATYPEMODULESDIR)\"" \
tools/build/Makefile.vars:310:	-DUPDATEMODULESDIR="\"$(UPDATEMODULESDIR)\""
[leebooil@leebooil-dev soletta]$ grep -nr LIBDIR *
config.log:77:LIBDIR = /usr/lib/
config.log:78:LIBDIR_RELATIVE = ${prefix}/lib/
Makefile.smallos:40:export PKG_CONFIG_LIBDIR=
pc/soletta.pc.in:3:libdir={{@LIBDIR_RELATIVE@}}
tools/build/Makefile.vars:92:export LIBDIR ?= $(PREFIX)/lib/
tools/build/Makefile.vars:98:export MODULESDIR := $(LIBDIR)$(PKGNAME)/modules/
tools/build/Makefile.vars:103:export PCDIR := $(LIBDIR)/pkgconfig/
tools/build/Makefile.vars:111:# _RELATIVE variants replaces $(PREFIX) with `${prefix}', $(LIBDIR) with `${libdir}' and so on, to be used in shell scripts and ".pc" files
tools/build/Makefile.vars:112:export LIBDIR_RELATIVE := $(subst $(patsubst %/,%,$(PREFIX)),$${prefix},$(LIBDIR))
tools/build/Makefile.vars:118:export MODULESDIR_RELATIVE := $(subst $(patsubst %/,%,$(LIBDIR)),$${libdir},$(MODULESDIR))
tools/build/Makefile.vars:137:build_libdir := $(build_sysroot)$(LIBDIR)

#### Board information

Host OS and architecture: Ubuntu 14.04.4 LTS

Target OS and platform: Linux 4.1, APL-i

#### Logs

barbieri, 2016-05-16:
@lblim did you try to provide `LIBDIR=/usr/lib64`? That should work, use when you specify your compiler and other sdk variables.

lblim, 2016-05-18:
Yes, it should work. I will try that. I assume this will be solved in the soletta recipe by providing something LIBDIR="${libdir}"

lblim, 2016-05-19:
An update, this problem only happens if multilib is enabled in yocto; if multilib is disabled the Soletta image will be installed at /usr/lib.

brunobottazzini, 2016-05-24:
@lblim, I tried to replicate the problem here so I would be able to investigate this problem further but, I was not able to replicate it (with and without multilib).

I tried setting target Machine to intel-corei7-64 but the build was successfully, maybe this is not enough?

Could you help us out with more information about the build system of yours so we can figure this out?

lblim, 2016-05-25:
@brunobottazzini , here is the configuration in local.conf that @chongyic used for our Yocto BSP. 

# Enable multi-libraries
require conf/multilib.conf
DEFAULTTUNE = "corei7-64"
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "corei7-32"

brunobottazzini, 2016-05-30:
@lblim, I could replicate the issue here... I'm working on a fix.

Thank you

brunobottazzini, 2016-05-30:
@lblim, After some tests, I've created a new PR to fix this issue: https://github.com/solettaproject/meta-soletta/pull/87

Could you please review and test it ? 

Best Regards,

brunobottazzini, 2016-06-01:
Hi @lblim, were you able to test the pr ?

Thanks....

lblim, 2016-06-02:
@chongyic Please update your test result to this issue.

Thanks

chongyic, 2016-06-03:
Hi @brunobottazzini  @lblim ,

The build looks fine.
Thanks.

brunobottazzini, 2016-06-03:
@chongyic, @lblim,

Thank you very much.

Patch merged and issue solved.

closed 2016-06-03 by brunobottazzini