solettaproject/soletta #2044: IIO node by default assume the IIO driver has buffer implemention and give warning message

lblim, 2016-05-16:
#### Issue Description

IIO node by default assume the IIO driver has buffer implemented, but not all IIO drivers have buffer implementation, therefore the lookup for buffer entry in sysfs may return warning at user space and also need to have buffer_size entry as -1 in the JSON file.

    } else {
        /* buffer_size == -1 means that user doesn't want to use the buffer */
        device->buffer_enabled = false;
        if (!set_buffer_enabled(device, false)) {
            SOL_WRN("Could not disable buffer for device%d", device->device_id);
        }
    }

#### Steps to reproduce
 1. Create FBP for IIO node without buffer implementation, for example BMP280, AK09911
 2. Run FBP with sol-flow-runner
 3. look for message at console

#### Board information

Host OS and architecture: Ubuntu 14.04.4 LTS

Target OS and platform: Linux 4.1 APL-i

barbieri, 2016-05-16:
@yongli3 can you look into this? otherwise let @edersondisouza know

yongli3, 2016-05-17:
@lblim I submitted a PR for this issue. For such "no-buffer" devices, If enable buffer reading in fbp files(buffer_size > -1), Soletta will output warning message and quit, and does not call the buffer disable API if there is no buffer/enable file

edersondisouza, 2016-06-06:
@lblim This issue should've been addressed, can you confirm it, please?

lblim, 2016-06-08:
@fulong82 will confirm this fix once the build issue is resolved

fulong82, 2016-06-08:
I have verified the latest code and no more warning message appear if the iio sensor does not have buffer expose.

edersondisouza, 2016-06-08:
@fulong82 Thanks, closing issue then.

closed 2016-06-08 by edersondisouza