solettaproject/soletta #2045: Instantiation of same IIO sensor node multiple times using create function shows error message in dmesg

lblim, 2016-05-16:
#### Issue Description

if instantiate same IIO sensor node multiple times using create function in FBP flow will give error message in dmesg. Though it is not harmful as all still function correctly, but this may be misleading if user read the dmesg. Should Soletta check for the node existence before trying to register it again?

[ 2121.439994] i2c i2c-6: new_device: Instantiated device bmp280 at 0x77
[ 2510.715439] i2c i2c-6: Failed to register i2c client bmp280 at 0x77 (-16)
[ 2612.374461] i2c i2c-6: Failed to register i2c client bmp280 at 0x77 (-16)
[ 4249.787058] i2c i2c-6: Failed to register i2c client bmp280 at 0x77 (-16)
[ 4300.489290] i2c i2c-6: Failed to register i2c client bmp280 at 0x77 (-16)
[ 4845.870106] i2c i2c-6: Failed to register i2c client bmp280 at 0x77 (-16)

root@intel-corei7-64:/opt/soletta-dev-app/repos/singlesession/testWebPress# vi sol-flow.json
{
  "$schema": "http://solettaproject.github.io/soletta/schemas/config.schema",
  "nodetypes": [
    {
      "name": "MyPress",
      "type": "iio/pressure",
      "options": {
        "iio_device": "create,i2c,pci0000:00/0000:00:17.2/i2c_designware.6,0x5d,lps25h",
        "buffer_size": 0
      }
    },
    {
      "name": "MyGyro",
      "type": "iio/gyroscope",
      "options": {
        "iio_device": "create,i2c,pci0000:00/0000:00:17.2/i2c_designware.6,0x69,bmg160",
        "buffer_size": 0
      }
    },
    {

#### Steps to reproduce

 1. Create FBP with IIO node (example shown above)
 2. Run and stop the fbp
 3. Repeat step 2 multiple times

#### Board information

Host OS and architecture: Ubuntu 14.04.4 LTS

Target OS and platform: Linux 4.1 APL-i

barbieri, 2016-05-16:
@lblim we're doing a work to keep a single instance of the IIO device in our node types (so we can have multiple kinds of node types for a single sensor, like luminosity and presence), which will allow this to work better.

However, consider that the `iio_device` parameter takes a space-separated list of arguments. Instead of using only "create,i2c...", which will try to create, you may add the actual device as precedence. Like in https://github.com/solettaproject/soletta/blob/master/src/samples/flow/iio/sol-flow-intel-minnow-max-linux_gt_3_17.json you can see `l3g4200d create,i2c,platform/80860F41:05,0x69,l3g4200d 0`, that will try first the device attached to module `l3g4200d`, if not there it will try `create,i2c,platform/80860F41:05,0x69,l3g4200d`, if not possible it will try `0` (the first iio device)

lblim, 2016-05-18:
OK, got you. The error goes away by adding the device name in front. Your explanation sounds clearer to me than the description in header file (https://github.com/solettaproject/soletta/blob/master/src/lib/io/include/sol-iio.h) on the usage of the first parameter about the device attached to module and the last parameter about the number of iio device.

closed 2016-05-18 by lblim