solettaproject/soletta #2058: Left shift formula exceeded boundary if iio channel->bits is 32 for iio buffer data reading

fulong82, 2016-05-19:
#### Issue Description

If IIO channel->bits is 32, the "channel->mask = (1 << channel->bits) - 1" will get zero due to exceeded the unsigned integer boundary and the left shift become left rotate.

Code located at file "src/lib/io/sol-iio.c", function "sol_iio_add_channel", inside if statement "device->buffer_enabled".

#### Steps to reproduce

 1. Get an IIO sensor driver that has 32bits data channel (example: drivers/iio/accel/hid-sensor-accel-3d.c)
 2. Configure the sol-flow.json to point to the iio device id and buffer_size = 0
 3. Run the src/samples/flow/iio/accelerate.fbp sample code

#### Example C code & its output
==Code==
#include <stdio.h>

void main (void)
{
	unsigned long long a = 32;
	unsigned long long b = 33;

	printf("shifted 1 left with variable a = 32: %016llX\n", (1 << a));
	printf("shifted 1 left with variable b = 33: %016llX\n", (1 << b));
	printf("shifted 1 left with immediate value 32 = %016llX\n", (1 << 32));
	printf("shifted 1 left with immediate value 33 = %016llX\n\n", (1 << 33));

	printf("shifted 1 left with variable a = 32: %016llX\n", ((unsigned long long)1 << a));
	printf("shifted 1 left with variable b = 33: %016llX\n", ((unsigned long long)1 << b));
	printf("shifted 1 left with immediate value = 32: %016llX\n", ((unsigned long long)1 << 32));
	printf("shifted 1 left with immediate value = 33: %016llX\n", ((unsigned long long)1 << 33));
}

==Output==
shifted 1 left with variable a = 32: 0000000000000001
shifted 1 left with variable b = 33: 0000000000000002
shifted 1 left with immediate value 32 = 0000000000000000
shifted 1 left with immediate value 33 = 0000000000000000

shifted 1 left with variable a = 32: 0000000100000000
shifted 1 left with variable b = 33: 0000000200000000
shifted 1 left with immediate value = 32: 0000000100000000
shifted 1 left with immediate value = 33: 0000000200000000

#### Soletta Logs (SOL_LOG_LEVEL = 5)
DBG: ./src/lib/io/sol-iio.c:964 sol_iio_add_channel() channel [in_accel_x] added. scale: 0.000009 - offset: 0 - storagebits: 32 - bits: 32 - **mask: 0**
... ...
DBG: ./src/lib/io/sol-iio.c:964 sol_iio_add_channel() channel [in_accel_y] added. scale: 0.000009 - offset: 0 - storagebits: 32 - bits: 32 - **mask: 0**
... ...
DBG: ./src/lib/io/sol-iio.c:964 sol_iio_add_channel() channel [in_accel_z] added. scale: 0.000009 - offset: 0 - storagebits: 32 - bits: 32 - **mask: 0**
DBG: ./src/lib/io/sol-iio.c:140 craft_filename_path() available=4096, used=46, path='/sys/bus/iio/devices/iio:device6/buffer/enable'
DBG:sol-flow ./src/lib/flow/sol-flow-static.c:232 connect_nodes() Making 1 connections.
#anon:6:17 (0.000000, 0.000000, 0.000000) (direction-vector)
#anon:6:17 (0.000000, 0.000000, 0.000000) (direction-vector)
#anon:6:17 (0.000000, 0.000000, 0.000000) (direction-vector)
#anon:6:17 (0.000000, 0.000000, 0.000000) (direction-vector)

 * Soletta version (top commit hash): tag v1_beta19

lblim, 2016-05-19:
@yongli3 This is the issue Wilson discussed with you that he caught on our platform with sensor hub.

barbieri, 2016-05-19:
when adding the cast, use `(uint64_t)` instead of `(unsigned long long)`

barbieri, 2016-05-19:
@yongli3 if you can send the patch, let me know. Otherwise I need someone to cover as @edersondisouza is on vacations :-)

yongli3, 2016-05-20:
@barbieri I submitted a PR #2071, please check on it

fulong82, 2016-06-08:
I have verified this and now the iio-hid sensors are no longer showing Zero data anymore.

closed 2016-06-09 by fulong82