solettaproject/soletta #2088: IIO sysfs absolute path does not support "/sys/bus/iio/devices/iio:deviceX/"

fulong82, 2016-05-24:
#### Issue Description

When using "/sys/bus/iio/devices/iio:deviceX/" absolute path in the json conffile, the function resolve_absolute_path() in /src/lib/io/sol-iio.c file will enter an infinite looping until the timeout happen eventually failed to run the fbp.

This issue may due to the assignment of "result.path" of that function inside the if statement "realpath(address, real_path)". Comparing this assignment with function resolve_i2c_path(), strongly believe the correct way is "**result.path = real_path;**" instead of "result.path = address;".

#### Steps to reproduce

 1. Create/modify the sol-flow.json file to use absolute path point to "/sys/bus/iio/devices/iio:deviceX/"
 2. Run the fbp
 3. Observe the error log

#### Logs

 * Soletta version (top commit hash): tag v1_beta19
 * Soletta log (SOL_LOG_LEVEL = 5): [Log](https://gist.github.com/fulong82/e989a2ad3d02665f3a994384b87c8e2d)

fulong82, 2016-05-24:
Hi @lblim, @yongli3 & @barbieri,

This is another issue while i testing with iio absolute path. I did raise my curious on the real_path vs address in function resolve_absolute_path() to @yongli3 last week before I hit into this issue. When looking into the log, it seems that my doubt is correct. Educate me if this is not the expected design.

Thanks & Regards,
Wilson

barbieri, 2016-05-24:
Your proposal looks correct. The issue is because you have a trailing `/` in your option, the `realpath()` will strip that, as well as any double `/` in the path (ie: `/sys//bus/...`).

If you confirm that works as expected, then just send us the patch and have your name in our codebase ;-) Remember to add the `Signed-off-by: ` line as mentioned in https://github.com/solettaproject/soletta/wiki/Contributing So you create a pull request with your patch and refer to this bug `Closes: #2088`.

fulong82, 2016-05-25:
Okay, will try and see. 1st time doing github code contribution. :-)

edersondisouza, 2016-06-06:
@fulong82 As you made a PR to fix this issue, I'd guess it is already fixed, right?

fulong82, 2016-06-07:
Okay, will rebuild with latest code to double confirm the fix.

fulong82, 2016-06-08:
My build is gated by issue #2136 or [#91](https://github.com/solettaproject/meta-soletta/issues/91).
Will continue the verification once the build issue is fixed.

fulong82, 2016-06-08:
Okay... I have verified and the latest code is able to support direct path to "/sys/bus/iio/devices/iio:deviceX/"

edersondisouza, 2016-06-08:
@fulong82 Thanks, closing issue then.

closed 2016-06-08 by edersondisouza