solettaproject/soletta #2095: IIO sensor lost decimal point data in soletta output

fulong82, 2016-05-25:
#### Issue Description

When testing with BMP280 direct IIO sensor, observed all data after decimal point are zero. This is due to in file "src/lib/io/sol-iio.c" function sol_iio_read_channel_value(), when it call function sol_util_read_file() to read from iio sysfs interface, it use the "SCNd64" scan type which is an integer instead of floating point data type.

Propose to use a double data type instead of "%" SCNd64 for the function input argument.

#### Steps to reproduce

 1. Configure the sol-flow.json to use bmp280 and buffer_size = -1
 2. Run the bmp280 fbp
 3. Observe the showing data which has all zero after the decimal point
 4. Compare the data by directly cat the iio sysfs interface

#### Logs

root# cat /sys/bus/iio/devices/iio\:device1/in_pressure_input
100**.717609375**
root# cat /sys/bus/iio/devices/iio\:device1/in_pressure_input
100**.718613281**
root# cat /sys/bus/iio/devices/iio\:device1/in_pressure_input
100**.719449218**
root# cat /sys/bus/iio/devices/iio\:device1/in_pressure_input
100**.720707031**
root# SOL_FLOW_MODULE_RESOLVER_CONFFILE=sol-flow2.json sol-fbp-runner ./testWebPress.fbp
WRN: ./src/lib/io/sol-iio.c:681 sol_iio_open() Could not disable buffer for device1
#anon:6:38 100**.000000** (float range)
#anon:6:38 100**.000000** (float range)
#anon:6:38 100**.000000** (float range)
#anon:6:38 100**.000000** (float range)
#anon:6:38 100**.000000** (float range)
#anon:6:38 100**.000000** (float range)
^C
root#

 * Soletta version (top commit hash): tag v1_beta19

barbieri, 2016-05-25:
@edersondisouza or @yongli3, if the channels can be specified as double-precision/floating-point, we may need to declare that when defining the channels.

Can we query this somehow or is it defined "out of band", like in some spec? This may result in a new nodetype option, which I'd like to avoid.

yongli3, 2016-05-26:
In the sys/../scan_elements folder, there are some items such as in_accel_x_type, maybe we can use them as the channel value type? Let me test it using a BMP280 device

yma11, 2016-05-26:
Hi @yongli3, I also find pressure sensor mpl115 has similar issue as bmp280, pls have a look.

edersondisouza, 2016-06-06:
@fulong82 This issue should've been addressed, can you confirm it, please?

fulong82, 2016-06-07:
Okay, will rebuild with latest code to double confirm the fix.

fulong82, 2016-06-08:
I have verified the latest code and is capable to show the data after the decimal point.

edersondisouza, 2016-06-08:
@fulong82 Thanks, closing issue then.

closed 2016-06-08 by edersondisouza