solettaproject/soletta #2101: Proposal: Add support of IIO sampling frequency to channel level besides device level

fulong82, 2016-05-26:
#### Task Description

Hi @barbieri, @edersondisouza & @yongli3,

Base on the current design, IIO sampling frequency supported in soletta only cater to IIO sensor driver that its sampling frequency is shared by all (Refer to `src/lib/io/sol-iio.c` function `iio_set_sampling_frequency()` implemetation).

Whereas, there are a lot IIO sensor drivers exposing its sampling frequency that is shared by channel type. Let's take the sensor list [Link](https://github.com/ostroproject/ostro-os/blob/master/doc/hardware/sensors.rst) that is claimed in Ostro as examples to which are using channel type sampling frequency.

Accelerometer:
- BMC150 series -- http://lxr.free-electrons.com/source/drivers/iio/accel/bmc150-accel-core.c#L983
- MMA7455 series -- http://lxr.free-electrons.com/source/drivers/iio/accel/mma7455_core.c#L210

Gyro:
- BMG160 -- http://lxr.free-electrons.com/source/drivers/iio/gyro/bmg160_core.c#L729

Magneto:
- HMC5843 series -- http://lxr.free-electrons.com/source/drivers/iio/magnetometer/hmc5843_core.c#L468
- MAG3110 series -- http://lxr.free-electrons.com/source/drivers/iio/magnetometer/mag3110.c#L277

So a proposal here is to add support for the sampling frequency that are shared by channel type in IIO sensor.

Thanks.

#### Dependencies

N/A

laykuanloon, 2016-06-28:
Hi @barbieri, @edersondisouza & @yongli3,

I have 2 sensor need this enhancement also.

Following is the sensors:
Accelerometer - kxcjk1013 
/sys/bus/iio/devices/iio:device0/in_accel_sampling_frequency

Magnetometer - mmc35240
/sys/bus/iio/devices/iio:device5/in_magn_sampling_frequency

lblim, 2016-06-28:
@barbieri @bdilly Can this be pulled in for v1? The override sampling frequency option for IIO node will not work for certain sensors if the issue is not resolved.

barbieri, 2016-06-28:
not for v1, it's frozen. For v2 yes :-)

I'll talk to @bdilly so we can make v2 provide all the IIO enhancements you've asked, they seem to be quite interesting and useful.

lblim, 2016-06-29:
ok, can you prioritize these enhancements in v2? We may want to pull in the version with all these fixes instead of v1, v2 will be too late to meet our release date.

edersondisouza, 2016-07-01:
@laykuanloon @lblim @fulong82  On those devices, sampling is not actually per channel, it's just that sysfs entry contains channel name on it? Do those devices (or some of them) actually expose different type of channels? If that's so, we need to take this information into consideration when changing IIO API. For instance (with not so careful thought), a simple solution would be to add a `const char *channel_type` entry on  `sol_iio_config`. Then, internally, when setting sampling frequency, we would try this path (including channel type name) on `iio_set_sampling_frequency` on `sol-iio.c`.
However, if its possible to define separate per channel (and not per channel type), we need to think more carefully.
If a device actually exposes different channel types and different frequency sampling per type, then, instead of `const char *channel_type`, we would need a `const char *channel_type[]` and iterate over it on `iio_set_sampling_frequency`. In this case, the future 'independent nodes per channel' (#1998) feature needs to be planned together with those changes, in order to get a well designed solution.

laykuanloon, 2016-07-04:
@edersondisouza, so far I only see the sampling frequency is per channel type.

fulong82, 2016-07-18:
> @edersondisouza commented 17 days ago
> If a device actually exposes different channel types and different frequency sampling per type, then, instead of const char *channel_type, we would need a const char *channel_type[] and iterate over it on iio_set_sampling_frequency. In this case, the future 'independent nodes per channel' (#1998) feature needs to be planned together with those changes, in order to get a well designed solution.

Hi @edersondisouza,

They are some iio sensor drivers do expose different channel type in one driver with its own channel type sampling frequency, for example:
1)[BMI160](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/tree/drivers/iio/imu/bmi160/bmi160_core.c#n437) set & get sampling frequency are differentiated by `chan->type` for its accelero & gyro channel type.
2) [KMX61](http://lxr.free-electrons.com/source/drivers/iio/imu/kmx61.c#L844) set & get sampling frequency are differentiated by `chan->address` for its accelero & magneto channel type. 

But I do not think we need to implement `const char *channel_type[]` array, as each channel type already considered a new sensor node in Soletta. And I believe in Soletta when a sensor node being created, its sampling frequency will be initialized. So that will taken care of each channel type sampling frequency.

Regards,
Wilson

yongli3, 2016-08-15:
I am using the HMC5843 for testing, it exposes in_magn_sampling_frequency

yongli3, 2016-09-19:
@fulong82 @fulong82 PR https://github.com/solettaproject/soletta/pull/2301 was merged, please check if  this issue is resolved

fulong82, 2016-09-27:
Hi @yongli3,

Sorry for the late response. Thanks for the implementation. I have verified this patch with iio-hid accelerometer, gyro & magneto sensor. But i would like to request add-on support to light sensor and pressure sensor as our POR requirements do support iio/light-sensor & iio/pressure-sensor as well.

Thanks & Regards,
Wilson

ceolin, 2016-10-03:
I think this ticket can be closed, @fulong82 could you confirm if all the points were implemented  ?

fulong82, 2016-10-04:
Yes. Thanks for the implementation.

closed 2016-10-04 by fulong82