solettaproject/soletta #2110: OIC: Response to server request cannot be constructed asynchronously

gabrielschulhof, 2016-05-30:
#### Issue Description
```C
/* server.c */
#include <stdio.h>
#include <string.h>
#include <sol-oic-server.h>
#include <soletta.h>

struct request_response {
	struct sol_oic_request *request;
	struct sol_oic_response *response;
};

static bool sendResponse(void *data) {
	struct request_response *context = (struct request_response *)data;
	sol_oic_server_send_response(context->request, context->response,
		SOL_COAP_RESPONSE_CODE_OK);
	free(context);
	return false;	
}

static int putRequest(void *data, struct sol_oic_request *request) {
	printf("putRequest\n");
	bool async = true;
	struct request_response *context = malloc(sizeof(struct request_response));

	context->request = request;
	context->response = sol_oic_server_response_new(request);

	struct sol_oic_map_reader *reader =
		sol_oic_server_request_get_reader(request);
    struct sol_oic_repr_field field;
    enum sol_oic_map_loop_status end_status;
    struct sol_oic_map_reader iterator = {0, 0, 0, 0, 0, 0};
    SOL_OIC_MAP_LOOP(reader, &field, &iterator, end_status) {
		if (!strcmp(field.key, "async")) {
			async = field.v_boolean;
		}
	}

	if (async) {
		sol_idle_add(sendResponse, context);
	} else {
		sendResponse(context);
	}
	return 0;
}

static void startup() {
	struct sol_oic_resource_type resourceType = {
		SOL_SET_API_VERSION(.api_version = SOL_OIC_RESOURCE_TYPE_API_VERSION,)
		.put = { .handle = putRequest },
		.resource_type = { .data = "core.light", .len = 10 },
		.interface = { .data = "oic.if.baseline", .len = 15 },
		.path = { .data = "/a/xyzzy", .len = 8 }
	};
	sol_oic_server_register_resource(&resourceType, NULL,
		SOL_OIC_FLAG_DISCOVERABLE | SOL_OIC_FLAG_ACTIVE);
}

static void shutdown() {}

SOL_MAIN_DEFAULT(startup, shutdown);
```
```C
/* client.c */
#include <stdio.h>
#include <string.h>
#include <sol-oic-client.h>
#include <soletta.h>

static struct sol_oic_pending *pending = NULL;

#define MIN(x, y) (((x) < (y)) ? (x) : (y))

void putRequestComplete(void *data, enum sol_coap_response_code response_code,
	struct sol_oic_client *cli, const struct sol_network_link_addr *address,
	const struct sol_oic_map_reader *payload) {
	printf("put request complete\n");
}

static bool resourceFound(void *nothing, struct sol_oic_client *client,
	struct sol_oic_resource *resource) {
	struct sol_oic_request *request = NULL;
	struct sol_oic_map_writer *writer = NULL;
	struct sol_oic_repr_field field;

	printf("resourceFound: %.*s\n", SOL_STR_SLICE_PRINT(resource->path));
	if (sol_argc() > 1 && !strncmp(sol_argv()[1], resource->path.data,
			MIN(strlen(sol_argv()[1]), resource->path.len))) {

		request = sol_oic_client_request_new(SOL_COAP_METHOD_PUT, resource);
		writer = sol_oic_client_request_get_writer(request);
		field.key = "async";
		field.v_boolean = !strcmp(sol_argv()[2], "async");
		field.type = SOL_OIC_REPR_TYPE_BOOLEAN;
		if (sol_oic_map_append(writer, &field)) {
			printf("Failed to encode payload field\n");
		}
		sol_oic_client_request(client, request, putRequestComplete, NULL);

		sol_oic_pending_cancel(pending);
		resource = NULL;
	}

	return !!resource;
}

static void startup() {

	if (sol_argc() < 3) {
		printf("Usage: client <server-href> <sync|async>\n");
		sol_quit();
		return;
	}

	struct sol_oic_client *client = sol_oic_client_new();
	struct sol_network_link_addr address = {
		.addr = {
			.in6 = { 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 }
		},
		.family = SOL_NETWORK_FAMILY_INET,
		.port = 5683
	};
	if (!sol_network_link_addr_from_str(&address, "224.0.1.187")) {
		fprintf(stderr, "Failed to retrieve address");
		return;
	}
	pending = sol_oic_client_find_resources(client, &address, "", "",
		resourceFound, NULL);
}

static void shutdown() {}

SOL_MAIN_DEFAULT(startup, shutdown);
```

#### Steps to reproduce

 1. Build the above two files to create binaries ```server``` resp. ```client```
 2. In one terminal, run ```./server```
 3. In another terminal, run ```./client /a/xyzzy async```

At this point, the server dies with a double free. If you run the client with anything other than ```async``` as the last parameter, the request completes successfully.
#### Board information
Linux x86_64 laptop

Host OS and architecture:
Linux x86_64 laptop

Target OS and platform:
Fedora 23

otaviobp, 2016-05-31:
Thanks for reporting. we were really having a double free. #2116 fixes this bug.

otaviobp, 2016-06-01:
Fixed.

closed 2016-06-01 by otaviobp