solettaproject/soletta #2114: IIO node type: default values for device configuration

lblim, 2016-05-31:
#### Task Description
There are a few options available for IIO nodes to change the configuration of the nodes such as buffer size and sampling frequency. However, there is no output port for these configurations. How can the values of these configuration be read out?

barbieri, 2016-05-31:
what's the use case to read these values? These are values you declare and thus know beforehand.

lblim, 2016-06-01:
Not really, some of the frequency range is per hardware spec and it is defined in the driver code. 

Besides this, I am thinking of the use cases such as:
- fine tuning sampling frequency at run time via machine learning or other method that based on the scenarios in event monitoring or preventive maintenance that the sampling frequency may be at lower rate and increase when the sensor data over a certain thresholds
- may be like gateway management app that will display all the configurations for a sensor node for display purpose or allow the user to change after display the default value

May be another node type can be introduced that take in parameter such as sensor model, iio device number etc and then return the configuration info and allow to change the configuration info for control path. The existing implementation is more for data path.

barbieri, 2016-06-01:
Even if defined in the hardware and driver, it's know beforehand when you write the FBP or application. What I believe we could do is to allow some keywords/constants to define highest/lowest frequency and let soletta translate that to an actual number. Even dynamic behavior you suggest I believe it's best done inside the node type itself.

The second point and the last idea about a new node type I find a bit out of scope for the FBP usage, as they are more targeted at the development of the FBP itself. It could be done in DevApp, a visual FBP editor or something like that, which could either use a local information source (works in Linux) or provided database (works for MCU/Zephyr). Then it would be used to create the actual FBP, doesn't need to be an FBP itself.

lblim, 2016-06-02:
We are still exploring on the application development model. Most of our use cases will be the gateway that requires to discover what sensors attached to the gateway and then populate the sensor metadata and display to the users so that the users can query the status, can change the configuration and can collect the sensor data and pass it to cloud or process locally. Looks like FBP and IIO node may not be the suitable model for this kind of gateway management application. In this case, low level C-API for IIO may need to be used instead or creating custom nodes.

The existing IIO  C-API does not support the read out of configurations also. I will submit another ticket to enable more C-API.

lblim, 2016-06-02:
@barbieri 

I think allow some keywords/constant to define the range of sampling frequency allowable would be helpful at FBP level, as there are a lot of sensors needs to follow the sensor spec for the sampling frequency available and scale available. I was thinking how can the IIO node can eliminate the need of the user to read the sensor spec or sensor driver code in order to know these information.

barbieri, 2016-06-03:
@lblim what would be the user logic? My point is that we should implement that ourselves. Such as allow some constants to map to "minimum"  or "maximum" frequencies, maybe "middle" (ie: -1 is unset, use what was already specified, -2 is minimum, -3 is maximum and -4 is middle).

But we also need to look this from a broader angle: these kinds of sensors do not automatically show in the system. Either they come in the ACPI/devicetree, or they are manually loaded, either by specifying the i2c command for Soletta, or doing an initial service at system startup.

Then my point is that the entity that defines them could generate the FBP as well. Say you have a shell script that creates the i2c devices at boot, then you generate a series of:
```sh
sensor-${sensorid}(iio/${sensorclass}:iio_device="${sensorid}") OUT -> IN http-${sensorid}(http-server/${http_type}:path="/sensors/${sensorid}")
```

Then all sensors would be exposed in HTTP at a path "/sensors/${sensorid}". A similar construct can be used to create an `index.json` with all loaded sensors.

Can we do a `iio/list-devices`  node and a `iio/device-info`? Sure, we can, we do that for power-supply, netctl... my point is if we would aggregate value or just confuse the user.

lblim, 2016-06-24:
yes, fbp is a high level programming; so the programming model may be different from low-level C-API programming model. So there will be trade-off between the configuration/customization that can be made by the user (plus the information get exposed to the users) and the user friendliness (higher level abstraction with lesser complexity etc).

I did acknowledge that these kind of sensors do not automatically show in the system, I think making it easier to be showed up to the application developer will be a value there. Application developer may not be necessary the person who create the platform or populate the system with sensors.

I will close this issue since that the configuration settings issue should be addressed by issue #2117.

closed 2016-06-24 by lblim