solettaproject/soletta #2115: IIO node type: options to override offset and scale

lblim, 2016-05-31:
#### Task Description
For IIO nodes, there are options to override offset and scale. Some of these fields may be just readable in IIO kernel driver. It seems that Soletta code will always try to override the offset value. For HID-IIO driver codes these fields are read only. May I know what IIO sensor drivers support that?

 static bool
 567 iio_set_channel_offset(struct sol_iio_channel *channel, int offset)
 568 {   
 569     char path[PATH_MAX], *pure_name = NULL;
 570     bool result = false;
 571     
 572     SOL_NULL_CHECK(channel, false);
 573     
 574     if (craft_filename_path(path, sizeof(path), CHANNEL_OFFSET_PATH,
 575         channel->device->device_id, channel->name)) {
 576 
 577         if (sol_util_write_file(path, "%d", offset) > 0) {
 578             channel->offset = offset;
 579             return true;
 580         }
 581     }
 582     
 583     /* If failed, try channel pure name */
 584     pure_name = channel_get_pure_name(channel);
 585     if (pure_name && craft_filename_path(path, sizeof(path), CHANNEL_OFFSET_PATH,
 586         channel->device->device_id, channel->name)) {
 587 
 588         result = sol_util_write_file(path, "%d", offset) > 0;
 589         if (result)
 590             channel->offset = offset;
 591     }
 592     
 593     if (!result)
 594         SOL_WRN("Could not set offset to %d on channel [%s] of device%d",
 595             offset, channel->name, channel->device->device_id);
 596     
 597     free(pure_name);
 598     
 599     return result;
 600 }

927     if (config->scale > -1)
 928         iio_set_channel_scale(channel, config->scale);
 929     else
 930         channel_get_scale(channel);
 931 
 932     if (config->use_custom_offset)
 933         iio_set_channel_offset(channel, config->offset);
 934     else
 935         channel_get_offset(channel);
 936

lblim, 2016-05-31:
@yongli3 , this is the issue that i ping u just now.

barbieri, 2016-05-31:
edited the comment to add codeblock formatting (triple backtick)

yongli3, 2016-06-01:
@lblim  I just checked the MPU6050 sensor I am using, the scale is writable:
root@intel-corei7-64:/sys/bus/i2c/devices/1-0068/iio:device0# cat in_accel_scale
0.000598
root@intel-corei7-64:/sys/bus/i2c/devices/1-0068/iio:device0# cat in_accel_scale_available
0.000598 0.001196 0.002392 0.004785
root@intel-corei7-64:/sys/bus/i2c/devices/1-0068/iio:device0# echo 0.001196 > in_accel_scale
root@intel-corei7-64:/sys/bus/i2c/devices/1-0068/iio:device0# cat name
mpu6050
root@intel-corei7-64:/sys/bus/i2c/devices/1-0068/iio:device0# cat in_accel_scale
0.001196

lblim, 2016-06-01:
@yongli3 

yes, there are some drivers support changing that, but not all. The implementation in IIO node at Soletta will update the scale regardless of whether it is writable. Error may be returned if not writable. For HID IIO case, it is read only.

yongli3, 2016-06-01:
@lblim Please let me know how you set the offset/scale in your fbp files. The default option is -1, it means fbp will use default offset/scale. If the HID IIO does not support scale/offset, please do not set the offset/scale in the FBP files

lblim, 2016-06-02:
Yes, I do aware of that condition, the reason I am asking this is coming from the user perspective. The user will need to know which IIO device/sensor actually support custom offset/scale by reading spec or driver codes. I was thinking is there a way that IIO node can hide this complexity from the users, user may know what can be customized by querying or getting the metadata of the sensor. The code proceed to set the offset or scale if it is requested from the users even the fields are readable only. Though it is harmless for now as the code didn't check the error status and give a warning message only.

yongli3, 2016-06-02:
I checked the sol_iio_add_channel API again, it seems to me that the config->use_custom_offset is always false. @edersondisouza maybe we need to change the logic?

edersondisouza, 2016-06-06:
@yongli3 `use_custom_offset` is an option of each node type. It must be set to define custom offset (or scale, for which there's a `use_custom_scale`). It's is so because there's no default value for those options.

edersondisouza, 2016-06-06:
@lblim Soletta currently expects that users know the device they are using, so such discovery capacities are not implemented (I'm not sure if it's possible on IIO, actually).

lblim, 2016-06-24:
ok, I agree that fbp and node type are to hide some complexity from the users and the users most likely will use the default value configured by the drivers. So for more advance users that would like to do customization on the device configuration etc, they will need to either write a custom node themselves or writing the application using Soletta IIO API ([#2117](https://github.com/solettaproject/soletta/issues/2117))

closed 2016-06-24 by lblim