solettaproject/soletta #2118: Soletta/RIOT-OS: Unable To Build Any Of The 'soletta-small-oses-samples'

mzcs, 2016-06-02:
#### Issue Description

Hi There :)

Assistance On The Following Issue Would Be Greatly Appreciated ! :)

I've Set-up a Soletta/RIOT-OS Basis, As Was Suggested In The 'Build instructions' Section Here : https://github.com/solettaproject/soletta-small-oses-samples

However, None Of The Samples There Build & I Suspect of Some Soletta/RIOT-OS Related Mis-Configuration :

It Seem That While Soletta, Being Configured as a RIOT-OS Package, Does Recognize RIOT-OS ( As Per The Soletta config.log ), It Opts To Do a Linux-OS Build Instead : Ultimately Resulting In a Build Failure (of Any Sample - Including a Dummy).

Also, For Some Reason, In Order For Soletta' Config. Process To Find Installed Package' Libs. - Such as 'ICU', 'OpenSSL' & Etc., I Had To Manually Set The 'PKG_CONFIG_PATH' Env. Variable.

#### Steps to reproduce

 1. Set-up a a Soletta/RIOT-OS Basis.
 2. Manually Set The 'PKG_CONFIG_PATH' Env. Var. .
 3. Try To Build Any 'soletta-small-oses-samples' - Or Even a [Dummy](https://github.com/solettaproject/soletta/files/295397/test123.c.txt).

#### Board information

Host OS and Architecture: Linux x86_64 ( Kernel v4.6.0 ), Gentoo Hardened Dist.

Target OS and Platform: RIOT-OS, 'Native' Port (For Testing)

#### Logs

 * Soletta version (top commit hash): 7190f2a ( 02/Jun/2016 )

 * .config: [.config](https://github.com/solettaproject/soletta/files/295375/default.config.txt)

 * config log: [config.log](https://github.com/solettaproject/soletta/files/295362/config.log.txt)

 * make log (V = 1): [screenlog.0](https://github.com/solettaproject/soletta/files/295364/screenlog.0.txt)

 * make strace log (strace -ff make &> strace.log): [strace.log.gz](https://github.com/solettaproject/soletta/files/295390/strace.log.txt.gz)

ibriano, 2016-06-02:
I never made much of an effort to make native work, since we have a couple of Samr21-xpro boards and we used those to test on RIOT with radio working. I pushed a config.riot for the soletta-basic sample in the soletta-small-oses-samples repository, and changed the order the OSes are checked by Kconfig in #2122.
With those, the sample builds for native, but it crashes when I try to run it. I'll look into that now.

About pkg-config, why do you want it to detect those? We usually make sure that when building for !Linux, pkg-config is not used, since almost invariably those will be dependencies used by the Linux port only.

mzcs, 2016-06-02:
Thanks a Bunch For Looking Into It ! :)

As Per pkg-config, I've Just Realized That It Makes No Sense To Integrate Linux-x86_64 Libs. Into a !Linux-x86_64 Build... & Kindly Pardon Me For The Late Ignition :)

ibriano, 2016-06-03:
I traced the crash I'm seeing to fprintf() not being nice with native, now I need to see if this is something to fix on our side or on RIOT.

ibriano, 2016-06-03:
Just sent a pull request for RIOT (https://github.com/RIOT-OS/RIOT/pull/5508) to add fprintf() to the syscalls supported by native, so Soletta (and any other external package, for that matter) doesn't crash when using it.
There's still a close to infinite number of warnings produced when building for native, most of them due to the use of the -pedantic flag for GCC, but at least the basic sample now builds *and* runs. Should be a start to better support the native target.

bdilly, 2016-07-13:
@mzcs @ibriano 
I believe the build issue is fixed, so I'm closing this ticket.
If new problems appear please open new tickets specific for each problem you face.
Regarding the small os samples, I believe most of them were configured for a specific board, so when trying to run them on "native" target some tweaking may be necessary.
Thanks

closed 2016-07-13 by bdilly