solettaproject/soletta #2125: OIC: Response to request cannot be transmitted asynchronously

gabrielschulhof, 2016-06-02:
#### Issue Description
If you run the client/server pair from #2110 in sync, the put request completes successfully, but if you run it in async, you get
```
WRN:sol-coap ./src/lib/comms/sol-coap.c:805 on_can_write() Could not send packet 347 to  (0): Success
```
on the server side.
#### Steps to reproduce

 1. Build the files pasted into #2110 
 2. Run the server in one terminal
 3. Run the client in another terminal with parameters ```/a/xyzzy async```

At this point, on the server you will see the above-mentioned warning and the client will fail to display the string ```put request complete```. If you repeat the above steps but replace ```async``` with ```sync``` then the put request completes.

otaviobp, 2016-06-03:
I tested this on top of master and I couldn't reproduce the problem. Both cases, async and sync, are working for me. Can you provide me with extra information? What branch have you used?

gabrielschulhof, 2016-06-03:
### The async case:
client:
```
INF: ./src/shared/sol-util-file.c:432 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
INF: ./src/shared/sol-util-file.c:432 sol_util_iterate_dir() Could not open dir [/usr/share//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:204 sol_platform_get_board_name() board name is unknown
DBG:sol-update ./src/lib/common/sol-update.c:106 get_first_module_on_dir() Invalid update module dir: /home/nix/iot/soletta/build/soletta_sysroot//usr/lib/soletta/modules//update/
DBG:sol-update ./src/lib/common/sol-update.c:138 load_update_module() No update module to load
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:223 sol_init() Soletta 0.0.1 on linux-4.4.9-300.fc23.x86_64 initialized
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:262 sol_run() run
DBG:sol-coap ./src/lib/comms/sol-coap.c:1855 sol_coap_server_new_full() New server 0x796fa0 on port 0
WRN: ./src/lib/comms/sol-socket.c:54 sol_socket_ip_new() DTLS is not enabled, secure socket is not possible
WRN:sol-coap ./src/lib/comms/sol-coap.c:1802 sol_coap_server_new_full() Could not create socket (38): Function not implemented
INF: ./src/lib/comms/sol-oic-client.c:1440 sol_oic_client_new() DTLS support not built-in, only making non-secure requests
DBG:sol-coap ./src/lib/comms/sol-coap.c:796 on_can_write() CoAP packet sent (outgoing_len=1, pending_len=1) -- payload of 20 bytes, buffer holding it with 32 bytes
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 31058, href: '/oic/res', type: 1, header_code: 1, query: ''}
DBG:sol-coap ./src/lib/comms/sol-coap.c:1649 on_can_read() pkt received and parsed successfully
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 31058, href: '', type: 1, header_code: 69, query: ''}
resourceFound: /oic/d
resourceFound: /oic/p
resourceFound: /a/xyzzy
DBG: ./src/lib/comms/sol-oic-client.c:148 _best_server_for_resource() Resource does not have secure flag and we do not have DTLS support (using port 5683)
DBG: ./src/lib/comms/sol-oic-client.c:1113 _resource_request() Sending CoAP packet through non-secure server (port 5683)
DBG:sol-coap ./src/lib/comms/sol-coap.c:1649 on_can_read() pkt received and parsed successfully
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 31058, href: '', type: 1, header_code: 69, query: ''}
DBG:sol-coap ./src/lib/comms/sol-coap.c:712 timeout_expired() waiting 2345 ms to re-try packet id 13926
DBG:sol-coap ./src/lib/comms/sol-coap.c:796 on_can_write() CoAP packet sent (outgoing_len=1, pending_len=1) -- payload of 32 bytes, buffer holding it with 128 bytes
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 13926, href: '/a/xyzzy', type: 0, header_code: 3, query: ''}
DBG:sol-coap ./src/lib/comms/sol-coap.c:632 timeout_cb() server 0x796fa0 retrying packet id 13926 to client ::ffff:192.168.1.45
DBG:sol-coap ./src/lib/comms/sol-coap.c:712 timeout_expired() waiting 4690 ms to re-try packet id 13926
DBG:sol-coap ./src/lib/comms/sol-coap.c:796 on_can_write() CoAP packet sent (outgoing_len=1, pending_len=1) -- payload of 32 bytes, buffer holding it with 128 bytes
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 13926, href: '/a/xyzzy', type: 0, header_code: 3, query: ''}
DBG:sol-coap ./src/lib/comms/sol-coap.c:632 timeout_cb() server 0x796fa0 retrying packet id 13926 to client ::ffff:192.168.1.45
DBG:sol-coap ./src/lib/comms/sol-coap.c:712 timeout_expired() waiting 9380 ms to re-try packet id 13926
DBG:sol-coap ./src/lib/comms/sol-coap.c:796 on_can_write() CoAP packet sent (outgoing_len=1, pending_len=1) -- payload of 32 bytes, buffer holding it with 128 bytes
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 13926, href: '/a/xyzzy', type: 0, header_code: 3, query: ''}
^CDBG:sol-mainloop ./src/lib/common/sol-mainloop-impl-posix.c:237 on_sig_quit() got signal 2, quit main loop...
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:286 sol_quit_with_code() quit with code 0
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:303 sol_shutdown() shutdown
DBG:sol-update ./src/lib/common/sol-update.c:175 sol_update_shutdown() Update shutdown
```

server:
```
INF: ./src/shared/sol-util-file.c:432 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
INF: ./src/shared/sol-util-file.c:432 sol_util_iterate_dir() Could not open dir [/usr/share//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:204 sol_platform_get_board_name() board name is unknown
DBG:sol-update ./src/lib/common/sol-update.c:106 get_first_module_on_dir() Invalid update module dir: /home/nix/iot/soletta/build/soletta_sysroot//usr/lib/soletta/modules//update/
DBG:sol-update ./src/lib/common/sol-update.c:138 load_update_module() No update module to load
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:223 sol_init() Soletta 0.0.1 on linux-4.4.9-300.fc23.x86_64 initialized
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:262 sol_run() run
DBG:sol-coap ./src/lib/comms/sol-coap.c:1855 sol_coap_server_new_full() New server 0x1906060 on port 5683
WRN: ./src/lib/comms/sol-socket.c:54 sol_socket_ip_new() DTLS is not enabled, secure socket is not possible
WRN:sol-coap ./src/lib/comms/sol-coap.c:1802 sol_coap_server_new_full() Could not create socket (38): Function not implemented
[... cut ...]
DBG:sol-coap ./src/lib/comms/sol-coap.c:2107 sol_coap_packet_debug() {id: 49930, href: '', type: 2, header_code: 64, query: ''}
^CDBG:sol-mainloop ./src/lib/common/sol-mainloop-impl-posix.c:237 on_sig_quit() got signal 2, quit main loop...
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:286 sol_quit_with_code() quit with code 0
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:303 sol_shutdown() shutdown
DBG:sol-update ./src/lib/common/sol-update.c:175 sol_update_shutdown() Update shutdown

```

gabrielschulhof, 2016-06-06:
@otaviobp I think [```outgoing->cliaddr```](https://github.com/solettaproject/soletta/blob/master/src/lib/comms/sol-coap.c#L785) is messed up if you transmit asynchronously.

gabrielschulhof, 2016-06-06:
@otaviobp I've dumped its contents as hex and when you run with ```sync```, it has the same contents during both discovery and the request, but if you run with ```async``` its contents changes when attempting to send the response.

gabrielschulhof, 2016-06-06:
@otaviobp We need to copy ```cliaddr``` into the structure, instead of just storing a pointer to it, because ```cliaddr``` is a stack variable inside ```on_can_read()``` in ```sol-coap.c```.

gabrielschulhof, 2016-06-06:
@otaviobp PTAL at #2125 :)

closed 2016-06-06 by gabrielschulhof