solettaproject/soletta #2136: bitbake failed using the latest version on Ubuntu 64 bit

yongli3, 2016-06-06:
#### Issue Description
I would like to test the latest soletta source code in Ostro OS. I modified the SRCREV to the latest("53a63ecfeac53451951c4d96871d25f9aceca79a") in soletta-git.bb file and bitbake soletta. But the config failed with the below error:
| tools/kconfig/conf  --oldconfig Kconfig
| make: tools/kconfig/conf: Command not found
| tools/kconfig/Makefile:84: recipe for target 'oldconfig' failed
| make: *** [oldconfig] Error 127
| ERROR: oe_runmake failed

The tools/kconfig/conf is linked to an incorrect path:/lib/ld-linux-x86-64.so.2
        linux-vdso.so.1 =>  (0x00007fff24ff3000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f454a514000)
        /lib/ld-linux-x86-64.so.2 => /lib64/ld-linux-x86-64.so.2 (0x0000559aadb5f000)
The correct should be:
ldd tools/kconfig/conf
        linux-vdso.so.1 =>  (0x00007ffd21158000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f20a3287000)
        /lib64/ld-linux-x86-64.so.2 (0x0000555cb0d5b000)

yongli3, 2016-06-08:
@fulong82 For this build break issue, are you using Minnowmax? There is a workaround:
copy /lib64/ld-linux-x86-64.so.2 to /lib/ld-linux-x86-64.so.2

fulong82, 2016-06-08:
Okay ... will try and see. Thanks.

fulong82, 2016-06-09:
Okay i have tried the workaround by creating a symbolic link into my ubuntu system `/lib/` location. But this workaround seem weird to me. Base on my understanding, whatever yocto recipes compilation, they will use the build & library environment created at `yocto_build/tmp/sysroots/`. Isn't this a problem when they look for the library or toolchain from the host system instead of the cross compiled environment?

Please educate me if soletta build is different. Thanks.

yongli3, 2016-06-13:
@fulong82 Are you building the soletta for 64 bit system ? During the "bitbake soletta", tools/kconfig/conf will be executed, it can use these libs provided by host system

fulong82, 2016-06-13:
Hi @yongli3 ,

Yes, i am building the soletta at 64 bit system. But base on my understanding to yocto build system, whatever recipe build will base on top of the tool chain and rootfs environment created at `yocto_build/tmp/sysroots/` folder instead of using your host system library where may build with different source like eglibc or uglibc instead of glibc.

yongli3, 2016-06-13:
@fulong82 I am using the Ostro OS for Minnowmax as the test system. The device and the host are using the same libs for this conf tool. If using different yocto building system, these libs maybe different

bdilly, 2016-07-18:
Hi @yongli3 ,

I believe that with https://github.com/solettaproject/meta-soletta/pull/99 merged now this issue is fixed.

If it's not the case, please re-open this ticket.

Thank you

closed 2016-07-18 by bdilly

yongli3, 2016-07-19:
Thanks @bdilly for your update. I applied PR #99 the compile is okay but bitbake failed on installing process. Below is the error message. I have to modify the .bb file, to change "libsoletta.so.0.0.1" to "libsoletta.so.0.0.1", is it caused by my local building system? 
| mv: cannot stat '/home/yongli/proj/ostro-os/build/tmp-glibc/work/i586-nlp-32-ostro-linux/soletta/1_beta19+gitAUTOINC+7dac94293d-r0/image//usr                            /lib/libsoletta.so.0.0.1': No such file or directory
| WARNING: /home/yongli/proj/ostro-os/build/tmp-glibc/work/i586-nlp-32-ostro-linux/soletta/1_beta19+gitAUTOINC+7dac94293d-r0/temp/run.do_instal                            l.23615:1 exit 1 from 'mv /home/yongli/proj/ostro-os/build/tmp-glibc/work/i586-nlp-32-ostro-linux/soletta/1_beta19+gitAUTOINC+7dac94293d-r0/ima                            ge//usr/lib/libsoletta.so.0.0.1 /home/yongli/proj/ostro-os/build/tmp-glibc/work/i586-nlp-32-ostro-linux/soletta/1_beta19+gitAUTOINC+7dac94293d-                            r0/image//usr/lib/libsoletta.so'
| ERROR: Function failed: do_install (log file is located at /home/yongli/proj/ostro-os/build/tmp-glibc/work/i586-nlp-32-ostro-linux/soletta/1_                            beta19+gitAUTOINC+7dac94293d-r0/temp/log.do_install.23615)