solettaproject/soletta #2154: Proposal: add wildcard characters support to IIO absolute path json input

fulong82, 2016-06-10:
#### Task Description

Hi @barbieri, @edersondisouza and @yongli3,

I would like to make a suggestion here to **add support of wildcard characters in IIO absolute path input**. Below are my points of why this is useful to users when they choose to use absolute path:

- iio device id (iio:deviceX) is dynamically assigned base on device load sequence: This may cause the absolute path `/sys/bus/i2c/devices/6-0077/iio:device0/` changed on the next boot if it is not being loaded 1st. So if wildcard character is supported, then there won't be an issue if the absolute path wrote in `/sys/bus/i2c/devices/6-0077/iio:device*/`. Understand that you may suggest using `i2c/X-YYYY` form instead, below bullet shows another possibility.

-  i2c bus id (i2c-X) may be changed: When user choose to enable more i2c buses (e.g. smbus, graphic port or usb-to-i2c device), the i2c bus id may be changed. So the `i2c/X-YYYY` form could not help whereas the absolute path input `/sys/bus/i2c/devices/*-0077/iio:device*/` will do.

Besides letting user changing the sol-flow.json file, a wildcard characters support could eliminate the troublesome work by just adding a `glob()` function call before the `realpath()` function call in the `sol-iio.c` file `resolve_absolute_path()` function code.

#### Dependencies
N/A

#### Example Code 
```
#include <limits.h>
#include <stdlib.h>
#include <glob.h>
#include <stdio.h>
 
void main (void)
{
        char path[PATH_MAX] = "/sys/bus/i2c/devices/*-0077/iio:device*/";
        char resolved_path[PATH_MAX];
        glob_t resolved_glob;
        int flags;

        flags = GLOB_MARK | GLOB_TILDE;
        glob(path, flags, NULL, &resolved_glob);
        printf("glob return path = %s\n", resolved_glob.gl_pathv[0]);

        realpath(resolved_glob.gl_pathv[0], resolved_path);
        printf("realpath resolved path = %s\n\n", resolved_path);

        globfree(&resolved_glob);
}
```

edersondisouza, 2016-06-10:
Hi,

But how would you assure that Soletta matches the right device? User (a 
developer using Soletta) should have more information about device 
setup, imho. Of course user may have to do some tests, but I'm not sure 
how would this be useful once they have a final setup. Am I missing 
something?

fulong82, 2016-06-13:
Hi @edersondisouza ,

Let me answer your question one by one.

1. If the glob() function return only one entry, then there is no doubt on the matching. If there are more matching, the glob_t structure would store all the matching entries in it as a list. Then it is depend on what approach is being chose. I could think of 3 ways for example (I personally prefer user friendly way):

 - _A user friendly method_: print out the whole list and let user input which entry he/she choose.

 - _Use 1st entry as default method_: pick the 1st entry and print a warning message to indicate how many matching is found and which is being chose.

 - _Refuse all method_: Reject all and print error message when more than one entry are found.

2. The linux system i2c bus id may change even in a close system where it is very unlikely but it has no guarantee. Let me take my system to do a small experiment and show you.
The experiment steps are: 1) boot up the system, 2) login, 3) check the i2c id, 4) reboot.
Below shows 2 logs where the 1st log is captured during 1st time boot up and the 2nd log is captured after few times (within 20x) rebooted. You could notice that the bus id has changed from 8 to 12 for i2c controller = `../../../devices/pci0000:00/0000:00:1f.1/i2c-X`.
May be the symptom is more easily being encountered when your system is having more i2c buses like mine -- 13 i2c buses.

 - 1st time boot up log

```
root@intel-corei7-64:~# ls -al /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 Sep 22 22:28 .
drwxr-xr-x 4 root root 0 Sep 22 22:28 ..
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-0 -> ../../../devices/pci0000:00/0000:00:16.0/i2c_designware.0/i2c-0
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-1 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1
lrwxrwxrwx 1 root root 0 Sep 22 23:39 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10
lrwxrwxrwx 1 root root 0 Sep 22 23:39 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-11
lrwxrwxrwx 1 root root 0 Sep 22 23:39 i2c-12 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-12
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-2 -> ../../../devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-2
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-3 -> ../../../devices/pci0000:00/0000:00:16.3/i2c_designware.3/i2c-3
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-4 -> ../../../devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-4
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-5 -> ../../../devices/pci0000:00/0000:00:17.1/i2c_designware.5/i2c-5
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-6 -> ../../../devices/pci0000:00/0000:00:17.2/i2c_designware.6/i2c-6
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-7 -> ../../../devices/pci0000:00/0000:00:17.3/i2c_designware.7/i2c-7
lrwxrwxrwx 1 root root 0 Sep 22 23:39 i2c-8 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-8
lrwxrwxrwx 1 root root 0 Sep 22 23:39 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-ALPS0001:00 -> ../../../devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-4/i2c-ALPS0001:00
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-ELAN221D:00 -> ../../../devices/pci0000:00/0000:00:16.3/i2c_designware.3/i2c-3/i2c-ELAN221D:00
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-IMPJ0003:00 -> ../../../devices/pci0000:00/0000:00:17.3/i2c_designware.7/i2c-7/i2c-IMPJ0003:00
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-INT3474:00 -> ../../../devices/pci0000:00/0000:00:16.0/i2c_designware.0/i2c-0/i2c-INT3474:00
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-INT34C1:00 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1/i2c-INT34C1:00
lrwxrwxrwx 1 root root 0 Sep 22 22:28 i2c-NXP1001:00 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1/i2c-NXP1001:00
root@intel-corei7-64:~#
```

- 
 - Xth time reboot log

```
root@intel-corei7-64:~# ls -al /sys/bus/i2c/devices/
total 0
drwxr-xr-x 2 root root 0 Sep 23 02:05 .
drwxr-xr-x 4 root root 0 Sep 23 02:05 ..
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-0 -> ../../../devices/pci0000:00/0000:00:16.0/i2c_designware.0/i2c-0
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-1 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1
lrwxrwxrwx 1 root root 0 Sep 23 02:08 i2c-10 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-10
lrwxrwxrwx 1 root root 0 Sep 23 02:08 i2c-11 -> ../../../devices/pci0000:00/0000:00:02.0/drm/card0/card0-DP-1/i2c-11
lrwxrwxrwx 1 root root 0 Sep 23 02:08 i2c-12 -> ../../../devices/pci0000:00/0000:00:1f.1/i2c-12
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-2 -> ../../../devices/pci0000:00/0000:00:16.2/i2c_designware.2/i2c-2
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-3 -> ../../../devices/pci0000:00/0000:00:16.3/i2c_designware.3/i2c-3
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-4 -> ../../../devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-4
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-5 -> ../../../devices/pci0000:00/0000:00:17.1/i2c_designware.5/i2c-5
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-6 -> ../../../devices/pci0000:00/0000:00:17.2/i2c_designware.6/i2c-6
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-7 -> ../../../devices/pci0000:00/0000:00:17.3/i2c_designware.7/i2c-7
lrwxrwxrwx 1 root root 0 Sep 23 02:08 i2c-8 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-8
lrwxrwxrwx 1 root root 0 Sep 23 02:08 i2c-9 -> ../../../devices/pci0000:00/0000:00:02.0/i2c-9
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-ALPS0001:00 -> ../../../devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-4/i2c-ALPS0001:00
[... cut ...]
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-IMPJ0003:00 -> ../../../devices/pci0000:00/0000:00:17.3/i2c_designware.7/i2c-7/i2c-IMPJ0003:00
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-INT3474:00 -> ../../../devices/pci0000:00/0000:00:16.0/i2c_designware.0/i2c-0/i2c-INT3474:00
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-INT34C1:00 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1/i2c-INT34C1:00
lrwxrwxrwx 1 root root 0 Sep 23 02:05 i2c-NXP1001:00 -> ../../../devices/pci0000:00/0000:00:16.1/i2c_designware.1/i2c-1/i2c-NXP1001:00
root@intel-corei7-64:~#
```

Thanks.

barbieri, 2016-06-29:
@fulong82 do you know if we could use another method to reliably find the device, like udev? you can check with `udevadm`.

While adding the `glob()` is indeed simple, I'm afraid it won't solve the problem.

fulong82, 2016-07-03:
Hi @barbieri, thanks for the input. But i am not familiar with `udevadm`, will study it more for better understanding on how can this solve the problem.

If we choose to use udevadm approach, then this created a dependency with udev. But i recall some system may choose not to use udev (for example quark, they use mdev).

Regards,
Wilson

yongli3, 2016-09-19:
@fulong82 Just FYI, the create_device_iter_cb is using /sys/devices/platform/ to detect the i2c bus number,     can you use it to fix the "bus id change" issue?

fulong82, 2016-09-27:
Hi @yongli3,

Sorry for the late response. About the `create_device_iter_cb()` function, are you referring using "create" in the iio device options? But i am more referring if the iio device driver is auto-loaded during system boot up by BIOS ACPI or Device Tree tables, then "create" may not work for this scenario.

Thanks & Regards,
Wilson

fulong82, 2016-09-29:
Hi @yongli3,

Besides from i2c bus sensors, here another example from iio-hid sensors that would require the wildcard characters support when using the absolute path option:

```
root@intel-corei7-64-cavs-hda:~# ls -al /sys/bus/iio/devices/
total 0
drwxr-xr-x 2 root root 0 Sep 29 09:32 .
drwxr-xr-x 4 root root 0 Sep 29 09:32 ..
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device0 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200083.17.auto/iio:device0
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device1 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200041.4.auto/iio:device1
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device2 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200031.5.auto/iio:device2
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device3 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200086.16.auto/iio:device3
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device4 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200073.11.auto/iio:device4
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device5 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200073.12.auto/iio:device5
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device6 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200076.8.auto/iio:device6
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio:device7 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-20008a.15.auto/iio:device7
lrwxrwxrwx 1 root root 0 Sep 29 10:15 iio_sysfs_trigger -> ../../../devices/iio_sysfs_trigger
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger0 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200083.17.auto/trigger0
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger1 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200041.4.auto/trigger1
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger2 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200031.5.auto/trigger2
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger3 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200086.16.auto/trigger3
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger4 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200073.11.auto/trigger4
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger5 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-200073.12.auto/trigger5
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger6 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0002/HID-SENSOR-200076.8.auto/trigger6
lrwxrwxrwx 1 root root 0 Sep 29 10:15 trigger7 -> ../../../devices/pci0000:00/0000:00:11.0/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/001D:8086:22D8.0003/HID-SENSOR-20008a.15.auto/trigger7
root@intel-corei7-64-cavs-hda:~#
```

Example of what will be changed:
- `/001D:8086:22D8.XXXX/`: XXXX number will be changed if there is new hid device added for next boot up
- `/HID-SENSOR-200041.X.auto/`: X will be changed each time boot up or unplug then re-plugin again
- `/{33AECD58-B679-4E54-9BD9-A04D34F0C226}/`: GUID unlikely will be changed, unless change different sensor hub

Thanks & Regards,
Wilson

Extra Info FYR: a list of all hid devices on my machine:
0003:046D:C315.0001 is usb keyboard, the rest all are integrated sensor hub. If you manage to find an external usb sensor hub, you may get the similar list as well.
```
root@intel-corei7-64-cavs-hda:~# ls -al /sys/bus/hid/devices/*/
/sys/bus/hid/devices/0003:046D:C315.0001/:
total 0
drwxr-xr-x 5 root root    0 Sep 29 09:32 .
drwxr-xr-x 5 root root    0 Sep 29 09:32 ..
-r--r--r-- 1 root root 4096 Sep 29 09:56 country
lrwxrwxrwx 1 root root    0 Sep 29 09:32 driver -> ../../../../../../../bus/hid/drivers/hid-generic
drwxr-xr-x 3 root root    0 Sep 29 09:32 hidraw
drwxr-xr-x 3 root root    0 Sep 29 09:32 input
-r--r--r-- 1 root root 4096 Sep 29 09:56 modalias
drwxr-xr-x 2 root root    0 Sep 29 09:56 power
-r--r--r-- 1 root root 4096 Sep 29 09:56 report_descriptor
lrwxrwxrwx 1 root root    0 Sep 29 09:32 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Sep 29 09:32 uevent
 
/sys/bus/hid/devices/001D:8086:22D8.0002/:
total 0
drwxr-xr-x 12 root root    0 Sep 29 09:32 .
drwxr-xr-x  7 root root    0 Sep 29 09:32 ..
drwxr-xr-x  3 root root    0 Sep 29 09:32 HID-SENSOR-200001.1.auto
drwxr-xr-x  5 root root    0 Sep 29 09:32 HID-SENSOR-200031.5.auto
drwxr-xr-x  5 root root    0 Sep 29 09:32 HID-SENSOR-200041.4.auto
drwxr-xr-x  5 root root    0 Sep 29 09:32 HID-SENSOR-200076.8.auto
drwxr-xr-x 27 root root    0 Sep 29 09:32 HID-SENSOR-2000e1.2.auto
drwxr-xr-x 27 root root    0 Sep 29 09:32 HID-SENSOR-2000e1.3.auto
drwxr-xr-x 25 root root    0 Sep 29 09:32 HID-SENSOR-2000e1.6.auto
drwxr-xr-x 25 root root    0 Sep 29 09:32 HID-SENSOR-2000e1.7.auto
[... cut ...]
drwxr-xr-x 3 root root    0 Sep 29 09:32 HID-SENSOR-2000b1.24.auto
drwxr-xr-x 3 root root    0 Sep 29 09:32 HID-SENSOR-2000b4.23.auto
drwxr-xr-x 3 root root    0 Sep 29 09:32 HID-SENSOR-2000c1.26.auto
lrwxrwxrwx 1 root root    0 Sep 29 09:56 driver -> ../../../../../bus/hid/drivers/hid-sensor-hub
-r--r--r-- 1 root root 4096 Sep 29 09:56 modalias
drwxr-xr-x 2 root root    0 Sep 29 09:56 power
lrwxrwxrwx 1 root root    0 Sep 29 09:32 subsystem -> ../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Sep 29 09:32 uevent
root@intel-corei7-64-cavs-hda:~#
```

closed 2016-11-25 by ceolin