solettaproject/soletta #2205: Proposal: Add support for iio mount matrix

laykuanloon, 2016-06-28:
#### Task Description
Please support iio mount matrix in soletta, I need this feature for the soletta sensor application(fbp or c api) can read out the data that following android orientation.

iio mount matrix is merge into mainline kernel recently (https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=dfc57732ad38f93ae6232a3b4e64fd077383a0f1).  

https://github.com/01org/android-iio-sensors-hal/blob/master/transform.c#L200 is the example of handling mount matrix.

Thanks.
#### Dependencies
N/A

barbieri, 2016-06-29:
I'm thinking about this and would likely solve it outside of the IIO:
 - `sol-util.h` to provide a matrix transformation code, for use in C code;
 - introduce a node type to convert matrix, not specific about IIO, but any direction vector

Is this enough? For sure we could add the matrix transformation to the IIO node using that `sol-util.h` function to be introduced, I'm just wondering if that is going to help if we provide a more generic `direction-vector/transform` node that can be used for all inputs, like other kinds of sensors, data from the internet, etc.

laykuanloon, 2016-06-30:
But this is iio mount matrix, is in iio driver. I think is better to put under iio layer.

laykuanloon, 2016-06-30:
@barbieri
Can include this feature in V1?

@fulong82 @lblim

barbieri, 2016-06-30:
ok, then it makes sense to have both the option and the node type to transform. The option will write directly to the kernel options.

As for v1, I don't want to commit to that in order to avoid delaying the v1 even further. However, if for bureaucracy purposes the v1 is delayed and we have the code ready, then it could go in v1.

laykuanloon, 2016-07-04:
FYI. The iio mount matrix sysfs entry is READ only.

lblim, 2016-09-13:
@laykuanloon please close the ticket

closed 2016-09-14 by laykuanloon