solettaproject/soletta #2233: iio ADD_CHANNEL logic cause scale 0

laykuanloon, 2016-07-13:
@yongli3 

Please check the following patch line 78 and 81 https://github.com/solettaproject/soletta/commit/38e2261dcd3ad7d6f057451acdcdb583671664a9
 +   if (base->use_device_default_scale) //should be if(!base->use_device_default_scale)
 +        channel_config.scale = scale;
 +
 +    if (base->use_device_default_offset) //should be if(!base->use_device_default_offset)
 +        channel_config.offset = offset;

yongli3, 2016-07-13:
Thanks @laykuanloon!

edersondisouza, 2016-07-13:
Patch that fixes it was merged, closing issue.

closed 2016-07-13 by edersondisouza

fulong82, 2016-07-14:
Hi @yongli3, @laykuanloon & @edersondisouza,

I would like to bring up a discussion here due to this issue. The code flow seem to be when there is an error on setting the scale/offset (no matter it is 0 or others value), Soletta only shows the warning and then continue rest of the functions but the data reading still using the scale/offset value input from user. So, if we could perform a read back after the warning and reset the value back to the original instead of user input value, i think this should be the proper handling code flow. What do you guys think?

```
root@intel-corei7-64-cavs-hda:/opt/soletta-dev-app/repos/singlesession/testfbp# ./runfbp.sh testWebMagneto.fbp sol-flow2.json
WRN: ./src/lib/io/sol-iio.c:559 iio_set_channel_scale() Could not set scale to 0.000000 on channel [in_magn_x] of device1
WRN: ./src/lib/io/sol-iio.c:559 iio_set_channel_scale() Could not set scale to 0.000000 on channel [in_magn_y] of device1
WRN: ./src/lib/io/sol-iio.c:559 iio_set_channel_scale() Could not set scale to 0.000000 on channel [in_magn_z] of device1
#anon:6:37 (-0.000000, -0.000000, 0.000000) (direction-vector)
#anon:6:37 (-0.000000, -0.000000, 0.000000) (direction-vector)
#anon:6:37 (-0.000000, -0.000000, 0.000000) (direction-vector)
```

Thanks & Regards,
Wilson

laykuanloon, 2016-07-14:
I think this is a good idea. Just prompt a message to inform user that the scale is revert to default value.

edersondisouza, 2016-07-14:
Hi @fulong82,

Actually, it doesn't use set value regardless of failing. If it fails to set, it keeps current value, with happens to be 0 (uninitialised). But I like the idea of using channel default in this case. It should be fairly simple to change logic on https://github.com/solettaproject/soletta/blob/master/src/lib/io/sol-iio.c#L941 and https://github.com/solettaproject/soletta/blob/master/src/lib/io/sol-iio.c#L946.
Please, if not doing it soon, open a issue for this task.

fulong82, 2016-07-15:
Hi @edersondisouza,

What i mean here is instead of
```
    if (config->scale > -1)
        iio_set_channel_scale(channel, config->scale);
    else
        channel_get_scale(channel);

    if (config->use_custom_offset)
        iio_set_channel_offset(channel, config->offset);
    else
        channel_get_offset(channel);
```
I would suggest 
```
    if (config->scale > -1)
        iio_set_channel_scale(channel, config->scale);

    channel_get_scale(channel);

    if (config->use_custom_offset)
        iio_set_channel_offset(channel, config->offset);

    channel_get_offset(channel);
```
I think it is a secure practice to read back whatever you have written to ensure the value, in case it has being changed/limited/conditioned by HW or such as signal interference and so on.

What do you think?

Regards,
Wilson

lblim, 2016-07-19:
@fulong82 Please open another ticket for the issue or suggestion as this ticket was closed.