solettaproject/soletta #2235: IIO: Activate device feature before read raw data

laykuanloon, 2016-07-13:
Task Description

Hi @barbieri , @edersondisouza , @yongli3 

There is a need to activate device feature before read raw data for some sensor. 

The apds9930 ambient light and proximity sensor which have following entry that need to be enable before read raw data.
/sys/.../iio:deviceX/in_proximity_en
/sys/.../iio:deviceX/in_illuminance_en

Following is the description and sysfs entry example for enable iio device feature:
What:		/sys/.../iio:deviceX/in_energy_en
What:		/sys/.../iio:deviceX/in_distance_en
What:		/sys/.../iio:deviceX/in_velocity_sqrt(x^2+y^2+z^2)_en
What:		/sys/.../iio:deviceX/in_steps_en
KernelVersion:	3.19
Contact:	linux-iio@vger.kernel.org
Description:
		Activates a device feature that runs in firmware/hardware.
		E.g. for steps: the pedometer saves power while not used;
		when activated, it will count the steps taken by the user in
		firmware and export them through in_steps_input.

Thanks.

Dependencies

N/A

barbieri, 2016-07-13:
For ease of use, we want to include this in our side in some way (to be thought), but these can be done at the "system level" with some script that setups the modules. Something analogous to "modprobe.conf" and "modules-autoload.conf" that would setup the sensors. I've discussed with @yongli3 team in the past as we already setup i2c devices. The outcome of that discussion was that it should be done with some devicetree or ACPI overlay that the board/system provides to state which extra devices were attached, doing the setup at a single place. The steps counter could be also configured at that level.

However, data such as steps counter is more like an application or runtime configuration, maybe you enable or disable that feature based on some need. In this case it may result in some extra IIO device ports in addition to node type options, so we can dynamically change them.

yongli3, 2016-07-26:
@barbieri After discussed with @laykuanloon , we think we can add a logic in soletta, when open a channel, to check if there is any in_???_en sysfs entities, if exits, then write 1 to enable it, before close the channel, write 0 to the entities to save power. How do you think about it ?

barbieri, 2016-07-27:
I think it's a good approach, we should test to see if any problems will appear, but overall looks good.

laykuanloon, 2016-08-01:
@barbieri and @yongli3 

I already submitted the PR. Please take a look
#2270

laykuanloon, 2016-08-11:
Close this since already merge.

closed 2016-08-11 by laykuanloon