solettaproject/soletta #2236: Proposal: IIO: support configfs interface to create hrtimer trigger

laykuanloon, 2016-07-13:
Hi @barbieri , @edersondisouza , @yongli3

Based on the current design, soletta only support create sysfs trigger in  `src/lib/io/sol-iio.c ` function `create_sysfs_trigger()` 

For sensor that use hrtimer trigger it require `mkdir /sys/kernel/config/iio/triggers/hrtimer-<trigger-name>` then only write the trigger (echo `"<trigger-name>" > /sys/.../iio:deviceX/trigger/current_trigger`)

Example:
mkdir /sys/kernel/config/iio/triggers/hrtimer-kxcjk1013-hr-dev0
echo "kxcjk1013-hr-dev0" > /sys/.../iio:deviceX/trigger/current_trigger

barbieri, 2016-07-13:
Like I wrote in #2235, these things should eventually go in a system level configuration, maybe due device tree or ACPI overlay, or just a shell script that executes during system startup that would do the actual I2C/IIO configuration.

But for ease of use, I believe we could do these for you. The tricky part is to define how to detect it's an hrtimer so `create_sysfs_trigger()` can do the `mkdir()`. Should we use some prefix, like `hrtimer:NAME`, in that case we can use `startswith("hrtimer:")` and proceed with the setup?

laykuanloon, 2016-07-13:
I think by 
Set the trigger name to "hrtimer-kxcjk1013-hr-dev0", 
In `create_sysfs_trigger` detect the trigger name prefix with "hrtimer-" then do the `mkdir()`
Then remove the trigger name prefix, echo "kxcjk1013-hr-dev0" into `current_trigger`

edersondisouza, 2016-07-13:
I'm not sure if Soletta should embrace this peculiarities: what comes next? A different kind of trigger will prompt yet another code on Soletta to deal with it?
On the bright side, hrtimer trigger appears to be a 'standard' one, as sysfs trigger.

laykuanloon, 2016-07-13:
I think is good to have it since hrtimer trigger is one of the iio driver in mainline.

barbieri, 2016-07-13:
I dislike to use "-" as separator, since it's the standard in actual names. So either use ":" or "/" as they are often used as path separators (as in `$PATH` or directories).

barbieri, 2016-07-13:
@edersondisouza indeed, it's a balance of ease of use.  IMO the end scenario is to provide the whole setup as outside of soletta scope. But then you download some example from the internet and needs lots of manual tweaking before you run your example, even if in the same board.

laykuanloon, 2016-07-14:
The separator can be ":" or "/".  But have to `mkdir /sys/kernel/config/iio/triggers/hrtimer-<trigger name>`. 

I try `mkdir mkdir /sys/kernel/config/iio/triggers/hrtimer:<trigger name>` , it will give error
[  520.490972] Invalid trigger type: hrtimer:<trigger name>

I try `mkdir /sys/kernel/config/iio/triggers/hrtimer/<trigger name>`, it will give error
mkdir: cannot create directory `/sys/kernel/config/iio/triggers/hrtimer': Invalid argument

barbieri, 2016-07-14:
that's expected, we can **create** the path from explicitly separated components (ie: `hrtimer:XXX` -> `/sys/kernel/config/iio/triggers/hrtimer-XXX`).

The existing IIO/soletta code was used and explained based on what goes in the `/sys/.../iio:deviceX/trigger/current_trigger`and that (unfortunately) doesn't match the one you must `mkdir` since it misses the `hrtimer-` prefix. So it's better to keep the string the same concept, but allow it to be prefixed with a "namespace" such as `hrtimer:`

laykuanloon, 2016-08-04:
@edersondisouza and @barbieri 
I created the PR.  #2273

yongli3, 2016-09-19:
@laykuanloon I noticed the hrtimer is included in Linux kernel 4.5+, which kernel version are you using?

laykuanloon, 2016-09-22:
@yongli3. I am using 3.14 kernel. the hrtimer implemtation is changed when the author upstream.

The diffrent in user space is the hrtimer trigger setup.
My kernel: mkdir /sys/kernel/config/iio/triggers/hrtimer-<trigger name>
In upstream: mkdir /sys/kernel/config/iio/triggers/hrtimer/<trigger name>

I can't test upstream kernel on my platform.

#2273 code can be reuse only change the mkdir part.

yongli3, 2016-09-23:
Thanks @laykuanloon I am using kernel 4.4 and it does not support the hrtimer trigger, let me switch to a higher version for the test. I think we can support both the old and new kernel

yongli3, 2016-09-29:
@laykuanloon I performed a test on my boards, I found I can create the "A:B" hrtimer. Can you double check if the ":" is okay on your boards? If not, can we use the same method as the sysfs triggers: using a random name as the trigger name?

laykuanloon, 2016-09-30:
@yongli3 My board only can `mkdir /sys/kernel/config/iio/triggers/hrtimer-<trigger name>`. Your hrtimer trigger is create under `/sys/kernel/config/iio/triggers/` folder?

yongli3, 2016-09-30:
It seems that different kernels have different behaviors. I cannot create folders in iio/triggers, but iio/triggers/hrtimer folder can support "aa:bb" format:
 
root@raspberrypi3:/sys/kernel/config/iio/triggers# ls -la
drwxr-xr-x    3 root     root             0 Sep 26 13:56 .
drwxr-xr-x    3 root     root             0 Sep 26 13:56 ..
drwxr-xr-x    4 root     root             0 Sep 27 10:47 hrtimer
root@raspberrypi3:/sys/kernel/config/iio/triggers# mkdir aaa
mkdir: can't create directory 'aaa': Operation not permitted

root@raspberrypi3:/sys/kernel/config/iio/triggers/hrtimer# ls -la
drwxr-xr-x    4 root     root             0 Sep 27 10:47 .
drwxr-xr-x    3 root     root             0 Sep 26 13:56 ..
drwxr-xr-x    2 root     root             0 Sep 27 10:41 1804289383
drwxr-xr-x    2 root     root             0 Sep 27 10:47 hrtimer:aaa
root@raspberrypi3:/sys/kernel/config/iio/triggers/hrtimer# ls -l /sys/bus/iio/devices/
iio:device0/       iio_sysfs_trigger/ trigger0/          trigger1/          trigger2/
root@raspberrypi3:/sys/kernel/config/iio/triggers/hrtimer# ls -l /sys/bus/iio/devices/
lrwxrwxrwx    1 root     root             0 Sep 26 13:59 iio:device0 -> ../../../devices/platform/soc/3f804000.i2c/i2c-1/1-001e/iio:device0
lrwxrwxrwx    1 root     root             0 Sep 26 13:55 iio_sysfs_trigger -> ../../../devices/iio_sysfs_trigger
lrwxrwxrwx    1 root     root             0 Sep 27 10:41 trigger0 -> ../../../devices/trigger0
lrwxrwxrwx    1 root     root             0 Sep 27 10:44 trigger1 -> ../../../devices/iio_sysfs_trigger/trigger1
lrwxrwxrwx    1 root     root             0 Sep 27 10:47 trigger2 -> ../../../devices/trigger2
root@raspberrypi3:/sys/kernel/config/iio/triggers/hrtimer# cat  /sys/bus/iio/devices/trigger2/name
hrtimer:aaa
root@raspberrypi3:/sys/kernel/config/iio/triggers/hrtimer# cat  /sys/bus/iio/devices/trigger0/name
1804289383

laykuanloon, 2016-09-30:
@yongli3 You are using upstream kernel right ? I think your method is correct. The logic to detect hrtimer is same (For e.g: user set  "iio_trigger_name": "hrtimer:kxcjk1013-hr-dev0"), only the mkdir part. I can generate patch on the mkdir portion for my release since my kernel hrtimer is not follow upstream kernel.

yongli3, 2016-09-30:
@laykuanloon I just submitted a PR #2322. It is based on your old #2273

laykuanloon, 2016-09-30:
@yongli3, I have manually merged your code and test. Thr hrtimer trigger is working for my kernel by make the following changes.

#if 1 // KERNEL 3.14
#define CONFIGFS_IIO_TRIGGERS_PATH "/sys/kernel/config/iio/triggers"
#define CONFIGFS_IIO_TRIGGER_PATH "/sys/kernel/config/iio/triggers/hrtimer-%s"
#else
#define CONFIGFS_IIO_TRIGGERS_PATH "/sys/kernel/config/iio/triggers/hrtimer"
#define CONFIGFS_IIO_TRIGGER_PATH "/sys/kernel/config/iio/triggers/hrtimer/%s"
#endif

yongli3, 2016-09-30:
@laykuanloon Thanks for test. Can you modify your kernel ? 
If not, we need to adjust the PR to compatible with both kernels

laykuanloon, 2016-09-30:
@yongli3. You can go ahead with your upstream implementation. I can generate patch on my platform.

peijiajames, 2016-10-24:
@laykuanloon the fix has been merged in. Could you please check the fix and close the task if no other issues? Thanks.

closed 2016-10-25 by laykuanloon