solettaproject/soletta #2244: Proposal: Add support for IIO Ambient Light which use in_intensity_both & in_intensity_ir as its channel data

fulong82, 2016-07-18:
@barbieri, @edersondisouza & @yongli3,

#### Task Description
Current Soletta only support IIO Ambient Light with `in_illuminance` processed channel type or `in_intensity_<COLORS>` raw channel type. I would like to make a request here to add support for `in_intensity_both` & `in_intensity_ir` raw channel type where the below ambient light sensor are using it:
1) [RPR0521](http://lxr.free-electrons.com/source/drivers/iio/light/rpr0521.c#L165)
2) [LTR501](http://lxr.free-electrons.com/source/drivers/iio/light/ltr501.c#L585)
3) [APDS9300](http://lxr.free-electrons.com/source/drivers/iio/light/apds9300.c#L373)
4) [TSL2563](http://lxr.free-electrons.com/source/drivers/iio/light/tsl2563.c#L553)
5) [HID ALS](http://lxr.free-electrons.com/source/drivers/iio/light/hid-sensor-als.c#L48)

#### Dependencies
N/A

yongli3, 2016-09-18:
@fulong82  Below is the sysfs nodes for TSL2563. Currently Soletta will read raw data from in_illuminance0_input, do you want to support both the _input and _raw data reading?

 cat name
tsl2563
root@raspberrypi3:/sys/bus/i2c/devices/1-0039/iio:device1# ls -l
-r--r--r--    1 root     root          4096 Sep 18 08:19 dev
drwxr-xr-x    2 root     root             0 Sep 18 08:19 events
-rw-r--r--    1 root     root          4096 Sep 18 08:19 in_illuminance0_input
-rw-r--r--    1 root     root          4096 Sep 18 08:19 in_intensity_both_calibscale
-rw-r--r--    1 root     root          4096 Sep 18 08:19 in_intensity_both_raw
-rw-r--r--    1 root     root          4096 Sep 18 08:19 in_intensity_ir_calibscale
-rw-r--r--    1 root     root          4096 Sep 18 08:19 in_intensity_ir_raw
-r--r--r--    1 root     root          4096 Sep 18 08:19 name
drwxr-xr-x    2 root     root             0 Sep 18 08:19 power
lrwxrwxrwx    1 root     root             0 Sep 18 08:19 subsystem -> >./../../../n./../../bus/iio
-rw-r--r--    1 root     root        ! 4096 Sep 18 08:18 uevent

fulong82, 2016-09-27:
Hi @yongli3,

Sorry for the late reply. For TSL2563 there are 2 interfaces which is illuminance and  intensity. Whereas, if you look at others like RPR0521, LTR501 as well as HID ALS, they only expose intensity. So for that case, i believe we need another note to support in_intensity_both and in_intensity_ir in Soletta in order to be able to use them in Soletta.

Thanks & Regards,
Wilson

yongli3, 2016-09-27:
@fulong82 please check on the PR #2318. I add a new option channel_name, you can set the channel name in .fbp files

edersondisouza, 2016-09-27:
Hi @fulong82 ,

`in_intensity_both` and `in_intensity_ir` contains which kind of information, respectively? If it's a different information than that from `in_intensity_<color>`, then they're worth new node types.

fulong82, 2016-09-27:
I don't mind to lump into `in_intensity_<color>`  since infra-red & visible light also can consider as light colors and they are all unit-less output.

edersondisouza, 2016-09-27:
But current output of color node type is an RGB packet, which doesn't make sense for IR or visible light. Maybe add two new integer output ports to current color node, IR and VISIBLE? Of course, channels wouldn't be mandatory for node, if at least one of them works...
Another approach would be two new nodes: iio/infra-red and iio/visible-light.

yongli3, 2016-09-27:
@edersondisouza @fulong82 if the channel_name(as a string option in .fbp files) is acceptable, we can use it to set different channels. For example, without the channel_name option, fbp will output the default channel in_illuminance0_input, set channel_name="in_intensity_ir_raw", fbp will output the IR_RAW data

fulong82, 2016-09-27:
Actually `in_intensity_both` mean visible + ir light according to sensor datasheets:
- [APDS9300](http://docs.avagotech.com/docs/AV02-1077EN)
- [TSL2563](http://ams.com/eng/content/download/250126/975581/142958)
- [LTR501](http://optoelectronics.liteon.com/upload/download/DS86-2012-0006/S_110_LTR-501ALS-01_PrelimDS_ver1[1].pdf)
- [RPR0521](http://rohmfs.rohm.com/en/products/databook/datasheet/opto/optical_sensor/opto_module/rpr-0521rs-e.pdf)

fulong82, 2016-09-27:
Hi @yongli3,

I am going to look into your suggestion. Once i understand how it work then i will response if it is fine of your suggestion. Thanks.

Regards,
Wilson

fulong82, 2016-09-29:
> For example, without the channel_name option, fbp will output the default channel in_illuminance0_input, set channel_name="in_intensity_ir_raw", fbp will output the IR_RAW data

Hi @yongli3,

Since the [#2318](https://github.com/solettaproject/soletta/pull/2318) implementation already changed to use index number, the example above is no longer valid. So for this ticket, are you suggesting doing the similar method which allows user to input modifier name? Or just adding an option in [color_create_channels()](https://github.com/solettaproject/soletta/blob/master/src/modules/flow/iio/nodes.c#L541) to allow selection of creating `in_intensity_<COLOR>`, `in_intensity_both` & `in_intensity_ir` ?

Thanks & Regards,
Wilson

yongli3, 2016-09-29:
I am thinking there are two methods:
1: Add a new node type named “light-intensity”, similar with the light/color, it will return in_intensity_both_raw and in_intensity_ir_raw
2: Enhance the current light node, it will return three values: illuminance, intensity_both and intensity_ir

peijiajames, 2016-10-27:
Leave a comment

peijiajames, 2016-10-27:
@fulong82 I see PR #2340 has been merged for this and you have verified it. Could you please close this task? Thanks.

fulong82, 2016-10-27:
Okay ... 👍

closed 2016-10-27 by fulong82