solettaproject/soletta #2245: MQTT: mosquitto_connect_async() not working when using mqtt sample c

laykuanloon, 2016-07-18:
@barbieri & @edersondisouza

I am using mosquitto 1.4.9 and soletta "b3544332560b526b6089a25891b3a80425a0864c" commit
I add mqtt log by apply 0001-mosquitto-add-mosquitto_log_callback_set.patch and capture the log

1) fbp mosquitto [https://github.com/solettaproject/soletta/blob/master/src/samples/flow/mqtt/mqtt_basic.fbp](url) is working.  I removed the sub mqtt client in fbp and only use pub mqtt client. 
mqtt-fbp.txt line 46 CONNECT, 47 PUBLISH, 48 CONNACK = only get CONNACK after first PUBLISH

2) C sample mosquitto [https://github.com/solettaproject/soletta/blob/master/src/samples/mqtt/mqtt-publish.c](url) is not working.
`mqtt-publish-sample.txt` line 8 CONNECT, after hang for 90 seconds then get error and reconnect then able to get CONNACK.

3) C sample mosquitto with 0002-mqtt-publish-do-sol_mqtt_publish-before-sol_run.patch is working. 
`mqtt-publish-sample-after-patch-publish-before-solrun.txt` line 8 CONNECT, 9 PUBLISH, 11 CONNACK mean mosquitto_connect_async() require PUBLISH then only can get CONNACK.

I have read [http://mosquitto.org/api/files/mosquitto-h.html#mosquitto_connect_async](url), it show If you use mosquitto_connect_async your client must use the threaded interface mosquitto_loop_start.  If you need to use mosquitto_loop, you must use mosquitto_connect to connect the client.

Please confirm [https://github.com/solettaproject/soletta/blob/master/src/lib/comms/sol-mqtt-impl-mosquitto.c#L411](url) is correct OR [https://github.com/solettaproject/soletta/blob/master/src/samples/mqtt/mqtt-publish.c#L62](url)
need PUBLISH without wait on_connect() callback being called.

[mqtt-fbp.txt](https://github.com/solettaproject/soletta/files/368883/mqtt-fbp.txt)
[mqtt-publish-sample.txt](https://github.com/solettaproject/soletta/files/368881/mqtt-publish-sample.txt)
[mqtt-publish-sample-after-patch-publish-before-solrun.txt](https://github.com/solettaproject/soletta/files/368882/mqtt-publish-sample-after-patch-publish-before-solrun.txt)

[mqtt problem.zip](https://github.com/solettaproject/soletta/files/368889/mqtt.problem.zip)

@fulong82 encountered different behavior compare to mine. I am connect to remote broker and @fulong82 connect to local broker. He get hang when run the C sample and will not reconnect like my 2) example.

cabelitos, 2016-07-18:
Hi @laykuanloon,

I found quite difficult to understand what's your problem. 
I tried to reproduce it here, however both samples are working fine for me. (fbp and C samples)

If I understand correctly, you're trying to setup two publishers, right? One will be the fbp and the another one will be the mqtt-publish.c, however you did not setup a subscriber! Note that you'll only receive/see the messages if you start a subscriber. In this case you will need to uncomment the Sub nodes (from the fbp) or run the mqtt-subsribe.c

If the problem still persists, try to run the mosquitto with the -v parameter. (`mosquitto -v`). If everything is working you should see something like:

```bash
1468876502: Received PUBLISH from mosq/s2wqEsixQjDBjxh>l6 (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
1468876502: Sending PUBLISH to mosq/tV\mVrQBiXIp6NKMFZ (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
1468876503: Received PUBLISH from mosq/s2wqEsixQjDBjxh>l6 (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
1468876503: Sending PUBLISH to mosq/tV\mVrQBiXIp6NKMFZ (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
1468876504: Received PUBLISH from mosq/s2wqEsixQjDBjxh>l6 (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
1468876504: Sending PUBLISH to mosq/tV\mVrQBiXIp6NKMFZ (d0, q0, r0, m0, 'soletta/mqtt/test', ... (12 bytes))
```
Note that the broker received the publish and then send it to a client.

Tip: Remeber to run the samples with SOL_LOG_LEVEL="DBG"

thanks

laykuanloon, 2016-07-19:
@cabelitos

I am running the applcation on Yocto + mosquitto 1.4.9 and soletta "b3544332560b526b6089a25891b3a80425a0864c" commit.

***I run one application at a time. 

You need to patch `0001-mosquitto-add-mosquitto_log_callback_set.patch.txt` to see mosquitto client log in soletta.

1) mosquitto fbp is working. Refer `mqtt-fbp.txt log` LINE 46 to 48 show that the mosquitto client receive CONNACK after PUBLISH
[mosquitto-publish.fbp.txt](https://github.com/solettaproject/soletta/files/370417/mosquitto-publish.fbp.txt)

2)mosquitto sample C app is not working because after sol_mqtt_connect() then it will wait for on_connect() callback being trigger which it does not happen. Refer `mqtt-publish-sample.txt` LINE 7 to 11
[https://github.com/solettaproject/soletta/blob/master/src/samples/mqtt/mqtt-publish.c](url)

3) is patch `0002-mqtt-publish-do-sol_mqtt_publish-before-sol_run.patch.txt` into mosquitto sample C app which I do sol_mqtt_publish() without wait the on_connect() callback being trigger. Refer `mqtt-publish-sample-after-patch-publish-before-solrun.txt` LINE 7 to 12. This is to prove mosquitto_connect_async() require PUBLISH first then only can get CONNACK.

laykuanloon, 2016-07-19:
I have tested the C sample with local mqtt broker and is working. Suspect is proxy problem.

bdilly, 2016-07-19:
I've done a few tests using local broker and remote broker, http://test.mosquitto.org/.
Everything seems to be working fine.
I'm also behind a proxy.

laykuanloon, 2016-07-20:
Weird, I am using the http://test.mosquitto.org/ test broker and bypass network proxy but still see same behavior.

# ./mqtt-publish test.mosquitto.org 1883 "soletta" "TEST MESSAGE"

INF:sol-platform ./src/lib/common/sol-platform-linux-common.c:947 sol_platform_impl_load_locales() The locale file (/etc/locale.conf) was not found in the system.
INF: ./src/lib/common/sol-util-file.c:433 sol_util_iterate_dir() Could not open dir [/etc//soletta/boards/] to iterate: No such file or directory
INF: ./src/lib/common/sol-util-file.c:433 sol_util_iterate_dir() Could not open dir [/usr/share//soletta/boards/] to iterate: No such file or directory
DBG:sol-platform ./src/lib/common/sol-platform.c:217 sol_platform_get_board_name() board name is unknown
WRN:sol-update ./src/lib/common/sol-update.c:99 get_first_module_on_dir() r (-1) < 0
DBG:sol-update ./src/lib/common/sol-update.c:138 load_update_module() No update module to load
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:223 sol_init() Soletta 0.0.0 on linux-3.14.55 initialized
WRN:sol-mqtt ./src/lib/comms/sol-mqtt-impl-mosquitto.c:349 my_log_callback() log callback Client mosq/asXUdP\ZDL[ml^oW^3 sending CONNECT
DBG:sol-mainloop ./src/lib/common/sol-mainloop.c:262 sol_run() run
WRN:sol-mqtt ./src/lib/comms/sol-mqtt-impl-mosquitto.c:167 sol_mqtt_event_loop() Unable to perform mqtt socket operation
INF: ./src/samples/mqtt/mqtt-publish.c:81 on_disconnect() Reconnecting...
WRN:sol-mqtt ./src/lib/comms/sol-mqtt-impl-mosquitto.c:349 my_log_callback() log callback Client mosq/asXUdP\ZDL[ml^oW^3 sending CONNECT
WRN:sol-mqtt ./src/lib/comms/sol-mqtt-impl-mosquitto.c:349 my_log_callback() log callback Client mosq/asXUdP\ZDL[ml^oW^3 received CONNACK
INF: ./src/samples/mqtt/mqtt-publish.c:45 publish_callback() 0: Sending Message.

laykuanloon, 2016-07-20:
@cabelitos and @bdilly

What is the mosquitto version use in your test environment ?

cabelitos, 2016-07-20:
@laykuanloon Hello,
I'm using Mosquitto 1.4.9

laykuanloon, 2016-07-21:
Ok, I am using mosquitto 1.4.9 also. I did debug into libmosquitto

In mosquitto_connect_async() function (libmosquitto code) will connect() and send(CONNECT) to broker. The problem is mosquitto didn't wait the SOCKET established then only write(CONNECT). 

AND soletta waiting the SOCKET become connected (Receive CONNACK from broker) which is not possible because the first write(CONNECT) is not send over ?

1) connect() get Operation now in progress. https://github.com/eclipse/mosquitto/blob/master/lib/net_mosq.c#L343
2) write() Resource temporarily unavailable https://github.com/eclipse/mosquitto/blob/master/lib/net_mosq.c#L746

Please read the mosquitto_connect_async() API documentation and confirm sol-mqtt-impl-mosquitto.c implementation is correct. 

> The API documentation said If you use mosquitto_connect_async your client must use the threaded interface mosquitto_loop_start. If you need to use mosquitto_loop, you must use mosquitto_connect to connect the client.

[mosquitto_connect_async()](https://mosquitto.org/api/files/mosquitto-h.html#mosquitto_connect_async
)

laykuanloon, 2016-07-22:
@cabelitos and @bdilly 

I found that connect() for non-blocking socket require to ppoll() with POLLOUT flag according to [http://stackoverflow.com/questions/28481993/non-blocking-socket-how-to-check-if-a-connection-was-successful?rq=1](url)

After I patch the following, then sol_mqtt_event_loop() can be trigger and continue do mosquitto_write() to complete the first write(CONNECT ). Do you agree on the patch?

diff --git a/src/lib/comms/sol-mqtt-impl-mosquitto.c b/src/lib/comms/sol-mqtt-impl-mosquitto.c
index d6d1a38..fea5ef5 100644
--- a/src/lib/comms/sol-mqtt-impl-mosquitto.c
+++ b/src/lib/comms/sol-mqtt-impl-mosquitto.c
@@ -419,7 +419,7 @@ sol_mqtt_connect(const struct sol_mqtt_config *config)
         goto error;
     }

-    mqtt->socket_watch = sol_fd_add(mqtt->socket_fd, SOL_FD_FLAGS_IN | SOL_FD_FLAGS_PRI,
+    mqtt->socket_watch = sol_fd_add(mqtt->socket_fd, SOL_FD_FLAGS_IN | SOL_FD_FLAGS_PRI | SOL_FD_FLAGS_OUT,
         sol_mqtt_event_loop, mqtt);
     SOL_NULL_CHECK_GOTO(mqtt->socket_watch, error);

laykuanloon, 2016-07-26:
@cabelitos and @bdilly 

Should I check in the patch ?

laykuanloon, 2016-07-27:
@edersondisouza and @barbieri 
Should I check in the patch ?

barbieri, 2016-07-27:
the patch should be harmless, so okay to apply. But that would be when you want to write some data without failling (ie: no errno `EAGAIN`).

@bdilly, @cabelitos and @laykuanloon please remember to check the transparent proxy configurations you may be using, things like `tsocks` and `redsocks`. Some of them are based on `LD_PRELOAD` environment variable and that can get going without people noticing them.

laykuanloon, 2016-07-28:
@bdilly, @cabelitos, @barbieri 
I have created the PR #2268 

@barbieri 
I don't know how to check the transparent proxy configuration. Do you have any test step I can follow?

I think the soletta mqtt does not support proxy now. I saw the mosquitto client example implement `mosquitto_socks5_set() ` which cannot find in soletta. 
[mosquitto_socks5_set()](https://github.com/eclipse/mosquitto/blob/master/client/client_shared.c#L702)

barbieri, 2016-07-28:
@laykuanloon it depends on the proxy. Most of the linux offers a transparent setting by using a custom library that wraps calls to `socket()`, `connect()` and the likes. That is, they set `LD_PRELOAD=somewapper.so`, and in `somewrapper.so` they provide these symbols, that will talk to the SOCKS server instead of traditional kernel `ioctl()` as done by libc.

Then check your environment variables if there is something suspect.

laykuanloon, 2016-08-08:
#2268 already merge. Close this. Thanks

closed 2016-08-08 by laykuanloon