solettaproject/soletta #2256: Proposal: IIO: Add read back to ensure the value written is being taken effect

fulong82, 2016-07-19:
@barbieri, @edersondisouza, @yongli3, @laykuanloon, @lblim

#### Task Description
The discussion is initiated from #2233. Open up this new ticket for task record as well as continue the discussion.

Here just a summary:
A secure coding practice would use the read back value instead of use the write value. A suggestion here:
Instead of
```
    if (config->scale > -1)
        iio_set_channel_scale(channel, config->scale);
    else
        channel_get_scale(channel);

    if (config->use_custom_offset)
        iio_set_channel_offset(channel, config->offset);
    else
        channel_get_offset(channel);
```
Do
```
    if (config->scale > -1)
        iio_set_channel_scale(channel, config->scale);

    channel_get_scale(channel);

    if (config->use_custom_offset)
        iio_set_channel_offset(channel, config->offset);

    channel_get_offset(channel);
```

Besides the scale and offset, suggest to apply this to sampling frequency as well to provide warning if the written value is not taken in effect.

Any different thought?

#### Dependencies
N/A

bdilly, 2016-07-19:
It seems this way you could make unnecessary reads.

I agree with you the way it is right now doesn't look secure, since it may fail to set channel scale (or offset).
But it could be done just checking for the return value of iio_set_channel_scale, don't you think?

Something like:
```C
bool r = false;
if (config->scale > -1)
    r = iio_set_channel_scale(channel, config->scale);

if (!r)
    channel_get_scale(channel);
```
But even get_scale() could fail and it would fallback to default value (1.0).

fulong82, 2016-07-20:
Hi @bdilly,

What i mean even a successful write, it is more secure to have a read back to ensure. Let take a scenario as an example: Some one is reading the raw data of the sensor and Soletta is doing a node initialization to this sensor. So the sensor HW may due to it already start sampling data and may ignore whatever write to some specific registers. In this case, use the read back value could prevent us using a wrong value if the HW does not take in effect.

Agree that most of the time this read back is un-necessary when thing just go perfectly. But it is a secure practice to take care corner case. Just to share some learning from secure coding training and open this discussion to seek better idea for it.

Thanks & Regards,
Wilson

edersondisouza, 2016-07-20:
Hi @fulong82,

> So the sensor HW may due to it already start sampling data and may ignore whatever write to some specific registers.

Is such a thing common? On a quick look through some iio drivers, they actually issue error when fail to set an attribute for any reason. If we see any driver doing wrong, then we may need to react. And things may go crazy here: what if a device needs some time before you can issue a read to confirm some write?
I'd go with @bdilly's approach - unless we already have cases like the one you stated. This would be a driver bug - it would be lying, after all - but sometimes we may need to workaround things...

yongli3, 2016-09-19:
@fulong82 @laykuanloon, @lblim any update/comments about this issue?

fulong82, 2016-09-27:
Hi @yongli3,

This is the remaining bug from #2233. I am just suggesting to make it common to all interface to have a read back after a write to ensure what you have written is taken in effect for safe practice.

Thanks & Regards,
Wilson

peijiajames, 2016-10-06:
Set agreement that strict return value check can satisfy the requirement.  @fulong82 @yongli3 please go with this direction and collaborate to close the task.

peijiajames, 2016-10-18:
@fulong82 fix is already merged by @ceolin  with #2325. Could you please check the fix and close the task? Thanks.

fulong82, 2016-10-18:
Hi @peijiajames, @yongli3 & @ceolin 

I have verified the merge in fix commit (https://github.com/solettaproject/soletta/commit/b87961255b24f52cf38f8e480b7f62eeab1ce706) which i found it different from (#2325). The write failed and fall back to driver default value issue is fixed. But there are unnecessary warning message showed even i did not mention to override the value.

**Case 1:** Intent to override scale but not the offset
```
root@intel-corei7-64:/opt/soletta-dev-app/repos/singlesession/test/SolettaWorkshop/Pressure# ../runfbp.sh Press.fbp sol-flow-hid.json
WRN: ./src/lib/io/sol-iio.c:590 iio_set_channel_scale() Could not set scale to 0.000000 on channel [in_pressure] of device5
WRN: ./src/lib/io/sol-iio.c:980 sol_iio_add_channel() Could not set scale (0) to the channel in_pressure
WRN: ./src/lib/io/sol-iio.c:989 sol_iio_add_channel() Could not set offset (0) to the channel in_pressure
pressure data = 100.517000 (float range) kPa
pressure data = 100.516000 (float range) kPa
```
sol-flow-hid.json configuration for pressure-sensor
```
{
      "name": "MyPress",
      "type": "iio/pressure-sensor",
      "options": {
        "iio_device": "press",
        "buffer_size": -1,
        "use_device_default_scale": false,
        "scale": 0
      }
}
```

**Case 2:** Do not plan to override scale as well as offset
```
root@intel-corei7-64:/opt/soletta-dev-app/repos/singlesession/test/SolettaWorkshop/Pressure# ../rubfbp.sh Press.fbp ../Mqtt/sol-flow-hid.json
WRN: ./src/lib/io/sol-iio.c:980 sol_iio_add_channel() Could not set scale (-1) to the channel in_pressure
WRN: ./src/lib/io/sol-iio.c:989 sol_iio_add_channel() Could not set offset (0) to the channel in_pressure
pressure data = 100.508000 (float range) kPa
pressure data = 100.510000 (float range) kPa
```
sol-flow-hid.json configuration for pressure-sensor
```
{
      "name": "MyPress",
      "type": "iio/pressure-sensor",
      "options": {
        "iio_device": "press",
        "buffer_size": -1
      }
}
```

So i think the `SOL_WRN("Could not set scale (%g) to the channel %s\n", config->scale, name);` and `SOL_WRN("Could not set offset (%d) to the channel %s\n", config->offset, name);` are not needed due to there already have warning message inside functions `iio_set_channel_scale()` & `iio_set_channel_offset()`.

Thanks & Regards,
Wilson

peijiajames, 2016-10-19:
oh, my bad. Actually the SOL_WARN message is suggested by me and I did not aware inner function already implemented the warning. @ceolin could you please check which warning message is needed? Thanks.

ceolin, 2016-10-19:
@peijiajames certainly. @fulong82 Thanks for catch this problem.

ceolin, 2016-10-20:
@fulong82 i've commited a fix for this, thanks. This issue can be closed ?

fulong82, 2016-10-21:
Sure thanks. 👍

closed 2016-10-21 by fulong82