solettaproject/soletta #2310: Denoise Filter Library for Sensors

lblim, 2016-09-13:
#### Task Description
Introduce denoise filter library into Soletta, it is needed for noisy sensors such as gyroscope and magnetometer
Type of denoise filters
- Mean / moving average filter
- Median filter

The filters and FBP node should be applied to all supported sensor types regardless 
Physical connection of sensors (either through sensor hub or direct connect)
Logical connection of sensors  (either through IIO or non-IIO)

lblim, 2016-09-14:
@laykuanloon @fulong82 @elvinongbl

barbieri, 2016-09-16:
did you check the buffer types? I recall @glima wrote them to address these

glima, 2016-09-16:
Yes,  if it's just mean/median you want, we already have support for that at least for integer types. Float could be added easily as well.

lblim, 2016-09-21:
@barbieri and @glima Thanks for pointing this out. Are you referring to https://github.com/solettaproject/soletta/blob/master/src/modules/flow/int/int.c

@laykuanloon Please take a look at the source code to see if it meets our requirements, we may extend on top of it if we need more than that. Also to expand for other data type such as float.

glima, 2016-09-21:
> @barbieri and @glima Thanks for pointing this out. Are you referring to https://github.com/solettaproject/soletta/blob/master/src/modules/flow/int/int.c

Yes, exactly (int/buffer node).

>
> @laykuanloon Please take a look at the source code to see if it meets our requirements, we may extend on top of it if we need more than that. Also to expand for other data type such as float.
>
> https://github.com/solettaproject/soletta/issues/2310#issuecomment-248482794

barbieri, 2016-11-18:
@lblim did that work for you?

barbieri, 2016-11-21:
@laykuanloon ping

laykuanloon, 2016-11-29:
@barbieri 
I implement denoise inside my sample app. [https://github.com/solettaproject/soletta/commit/45117b9b9951204bbb2c8c7d6709f76d0e0d4a30](url)

Is better to have a denoise that can work for C and sol_direction_vector data type. What do you think?

barbieri, 2016-11-29:
@lblim I agree, I'll propose this as a project, thanks.

In this project we will also propose denoise based on redundancy, like done in drones, one can use another sensor at different sampling rate, driver and mount, then use them to provide the same information and use the redundancy to reduce noise.

In addition to traditional buffer + function filtering as done before.