solettaproject/soletta #2315: Raspberry Pi 3 (Host & Target being the same) - make install - failed

mkharibalaji, 2016-09-23:
#### Task Description
Running 'make install' within Raspberry Pi 3 command line,the build is failing.

#### Dependencies

#### Issue Description
pi@raspberrypi:~/soletta $ make install
     GEN   ./build/soletta_sysroot/usr/share//soletta//flow//aliases//50-default.json
Alias file saved at ./build/soletta_sysroot/usr/share//soletta//flow//aliases//50-default.json
     GEN   include/generated/sol_config.h
*
* Restart config...
*
*
* Metatypes
*
JavaScript flow metatype (FLOW_METATYPE_JAVASCRIPT) [M/n/y/?] (NEW) y
*
* Node Types
*
Node type: accelerometer (FLOW_NODE_TYPE_ACCELEROMETER) [N/m/y] n
Node type: aio (FLOW_NODE_TYPE_AIO) [Y/n/m] y
Node type: am2315 (FLOW_NODE_TYPE_AM2315) [M/n/y] m
Node type: app (FLOW_NODE_TYPE_APP) [Y/n/m] y
Node type: boolean (FLOW_NODE_TYPE_BOOLEAN) [Y/n/m] y
Node type: byte (FLOW_NODE_TYPE_BYTE) [Y/n/m] y
Node type: calamari (FLOW_NODE_TYPE_CALAMARI) [M/n/y] m
Node type: color (FLOW_NODE_TYPE_COLOR) [Y/n/m] y
Node type: compass (FLOW_NODE_TYPE_COMPASS) [M/n/y] m
Node type: console (FLOW_NODE_TYPE_CONSOLE) [Y/n/m] y
Node type: constant (FLOW_NODE_TYPE_CONSTANT) [Y/n/m] y
Node type: converter (FLOW_NODE_TYPE_CONVERTER) [Y/n/m/?] y
Node type: evdev (FLOW_NODE_TYPE_EVDEV) [M/n/y] m
Node type: file (FLOW_NODE_TYPE_FILE) [M/n/y] m
Node type: filter-repeated (FLOW_NODE_TYPE_FILTER_REPEATED) [Y/n/m] y
Node type: float (FLOW_NODE_TYPE_FLOAT) [Y/n/m] y
Node type: form (FLOW_NODE_TYPE_FORM) [M/n/y/?] m
Node type: format (FLOW_NODE_TYPE_FORMAT) [M/n/y/?] m
Node type: gpio (FLOW_NODE_TYPE_GPIO) [Y/n/m] y
Node type: grove (FLOW_NODE_TYPE_GROVE) [M/n/y] m
Node type: gyroscope (FLOW_NODE_TYPE_GYROSCOPE) [N/m/y] n
Node type: hc-sr04 (FLOW_NODE_TYPE_HC_SR04) [N/m/y] n
Node type: iio (FLOW_NODE_TYPE_IIO) [M/n/y] m
Node type: int (FLOW_NODE_TYPE_INT) [Y/n/m] y
Node type: jhd1313m1 (FLOW_NODE_TYPE_JHD1313M1) [M/n/y] m
Node type: json (FLOW_NODE_TYPE_JSON) [M/n/y/?] m
Node type: keyboard (FLOW_NODE_TYPE_KEYBOARD) [M/n/y] m
Node type: led 7 segments (FLOW_NODE_TYPE_LED_7SEG) [Y/n/m] y
Node type: led-strip (FLOW_NODE_TYPE_LED_STRIP) [M/n/y] m
Node type: lsm9ds0 (FLOW_NODE_TYPE_LSM9DS0) [N/m/y] n
Node type: magnetometer (FLOW_NODE_TYPE_MAGNETOMETER) [N/m/y] n
Node type: max31855 (FLOW_NODE_TYPE_MAX31855) [N/m/y] n
Node type: network (FLOW_NODE_TYPE_NETWORK) [M/n/y] m
Node type: oic (FLOW_NODE_TYPE_OIC) [M/n/y] (NEW) y
Node type: persistence (FLOW_NODE_TYPE_PERSISTENCE) [M/n/y] m
Node type: piezo-speaker (FLOW_NODE_TYPE_PIEZO_SPEAKER) [M/n/y] m
Node type: platform (FLOW_NODE_TYPE_PLATFORM) [Y/n/m] y
Node type: power-supply (FLOW_NODE_TYPE_POWER_SUPPLY) [M/n/y/?] m
Node type: process (FLOW_NODE_TYPE_PROCESS) [M/n/y] m
Node type: pwm (FLOW_NODE_TYPE_PWM) [Y/n/m] y
Node type: random (FLOW_NODE_TYPE_RANDOM) [Y/n/m] y
Node type: robotics (FLOW_NODE_TYPE_ROBOTICS) [N/m/y/?] n
Node type: servo-motor (FLOW_NODE_TYPE_SERVO_MOTOR) [M/n/y] m
Node type: si114x (FLOW_NODE_TYPE_SI114X) [N/m/y] n
Node type: string (FLOW_NODE_TYPE_STRING) [Y/n/m/?] y
Node type: stts751 (FLOW_NODE_TYPE_STTS751) [N/m/y] n
Node type: switcher (FLOW_NODE_TYPE_SWITCHER) [Y/n/m] y
Node type: temperature (FLOW_NODE_TYPE_TEMPERATURE) [Y/n/m] y
Node type: test (FLOW_NODE_TYPE_TEST) [M/n/y] m
Node type: timer (FLOW_NODE_TYPE_TIMER) [Y/n/m] y
[... 245 lines cut ...]
#### Steps to reproduce

 1. Boot up Raspberry Pi 3
 2. Clone Git repo of soletta
 3. make thirdparty-update
 4. make alldefconfig
 5. make -j
 6. make install

#### Board information

Host OS and architecture: Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

Target OS and platform: Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

#### Logs

 * Soletta version (top commit hash): top commit
 * .config: link
 * config log: link
 * make log (V = 1): link
 * make strace log (strace -ff make &> strace.log): link
 * Soletta log (SOL_LOG_LEVEL = 5): link
 * strace log (strace -ff sol-fbp-runner test.fbp &> strace.log): link
 * valgrind log: link

yongli3, 2016-09-24:
I just performed the native build on Pi3, the result is sucessed. I am using " make alldefconfig; make". It seems to me that some dependency libs are missing ?

mkharibalaji, 2016-09-24:
Hi yongli,
          Thanks for your quick response.But I am getting error when i do "make install".Am I missing anything ?

 AR build/stage/shared/libsolettashared.a LD build/soletta_sysroot/usr/lib/libsoletta.so In function ‘__fread_alias’, inlined from ‘sol_memmap_impl_write_raw’ at ./src/lib/io/sol-memmap-storage-impl-linux.c:135:15: /usr/include/arm-linux-gnueabihf/bits/stdio2.h:290:2: warning: call to ‘__fread_chk_warn’ declared with attribute warning: fread called with bigger size * nmemb than length of destination buffer return __fread_chk (__ptr, __bos0 (__ptr), __size, __n, __stream); ^ /tmp/ccjHvC1j.ltrans7.ltrans.o: In function sol_lwm2m_security_add_full.lto_priv.1216': ccjHvC1j.ltrans7.o:(.text+0x30e8): undefined reference tosol_socket_dtls_set_credentials_callbacks' ccjHvC1j.ltrans7.o:(.text+0x3180): undefined reference to sol_socket_dtls_set_credentials_callbacks' /tmp/ccjHvC1j.ltrans9.ltrans.o: In functionsol_socket_ip_new': ccjHvC1j.ltrans9.o:(.text+0x44c8): undefined reference to sol_socket_default_dtls_new' /tmp/ccjHvC1j.ltrans24.ltrans.o: In functionsol_shutdown': ccjHvC1j.ltrans24.o:(.text+0x2a30): undefined reference to sol_oic_server_shutdown' /tmp/ccjHvC1j.ltrans24.ltrans.o: In functionsol_init': ccjHvC1j.ltrans24.o:(.text+0x33ec): undefined reference to `sol_oic_server_shutdown' collect2: error: ld returned 1 exit status tools/build/Makefile.rules:399: recipe for target 'build/soletta_sysroot/usr/lib/libsoletta.so' failed make: *** [build/soletta_sysroot/usr/lib/libsoletta.so] Error 1

Can you please help me out....

mkharibalaji, 2016-09-24:
Or could you please help me the things for the "y or n" options while doing make install such that it properly installs....

yongli3, 2016-09-24:
@mkharibalaji When I execute the "make alldefconfig" command, but I did not get these selection options.

mkharibalaji, 2016-09-24:
@yongli3 When I execute "make install" i am getting the error and the selection options.

ceolin, 2016-09-26:
Hi @mkharibalaji thanks for reporting the problem, i'll check it (have some guesses about the problem). Could you please run `make reconf` -> `make` -> `make install` and tell me whether the problem persists or not ?