solettaproject/soletta #2316: Make failed on NFS volume

yongli3, 2016-09-23:
I performed the " make alldefconfig; make" commands on a nfs mounted volume, it seems that the make process enter into a dead loop in "generates these .o.dep files"

yongli3, 2016-09-26:
It seems that the lin 11 in tools/build/Makefile.common has problems. the target is depending on the DEPENDENCY_CACHE file, but the .py command will update the DEPENDENCY_CACHE file, make tool will re-generate Makefile.gen file again and again

glima, 2016-09-26:
> It seems that the lin 11 in tools/build/Makefile.common has problems. the target is depending on the DEPENDENCY_CACHE file, but the .py command will update the DEPENDENCY_CACHE file, make tool will re-generate Makefile.gen file again and again

It looks a bit odd, indeed, but I wonder how haven't we hit this before.

> 
> https://github.com/solettaproject/soletta/issues/2316#issuecomment-249490244

yongli3, 2016-09-27:
@glima I did not encounter this issue before. But when I native build Soletta using Pi3/Minnowmax boards, the problem occurs...