solettaproject/soletta #2329: FBP as standalone project

dumblob, 2016-10-13:
#### Task Description

The FBP engine is already so powerful and comprehensive, that it would make sense to separate it from Soletta and develop it as a standalone product. There are plenty of use cases for a flow-based programming engine backed by very fast C implementation (generated source code).

Are there any such plans? Now it's complicated to use it without Soletta (because of tight coupling with Soletta headers etc.).

barbieri, 2016-10-16:
The FBP core relies on some primitives from soletta, from basic data types to main loop for its own signal dispatching.

That said, one can use soletta as "guest" library, like currently done with node.js bindings. In this case you can basically integrate its own main loop with something else without any changes to soletta itself (https://github.com/solettaproject/soletta/blob/master/src/lib/common/include/sol-mainloop.h#L846)

In this case it will only pull in the minimal stuff from soletta, unless you use some node type that happens to use other subsystems (ie: http-server would require soletta's wrapper around libmicrohttpd).

Of course, if you plan to recompile soletta, then you have even more control. One option is to use its kconfig to disable all but FBP, this will effectively not compile code for stuff you do not want. Another is to provide your own "main loop" implementation, currently on linux we provide one for Glib... 

Remember that the main loop is essential as every "packet sent" and its corresponding delivery is done in the "main thread", which is the one that runs the main loop. Each node type may wish to execute in a different thread or even process, but should deliver the results in the main thread. Of course, the delivery will happen in that thread as well -- NOTE: main thread is the one where soletta runs, it may not be a real main thread of your application as you may opt to run soletta itself on a secondary thread of your application.

Note that code-wise we do not bring too much overhead if things are not used. As you can see we run on micro-controllers (MCU), doing FBP + OIC + CoAP + CBOR + UDP in under 100kb, that includes the whole kernel and user application generated from FBP text file.

Do you have a more specific use case or need? Then we could advise you better :-)

dumblob, 2016-10-16:
Thank you @barbieri for all the important information. I think it's enough for now. I didn't expect the resulting binary being so small after all (that's amazing and is exactly what I need), nor the code being so well "ifdefed" for compile-time options of what to depend on etc.

>Do you have a more specific use case or need? Then we could advise you better :-)

I have 2 use cases (both for the same project). The first one is compilation to JS as a test/simulation back-end for a FBP web editor. And the second one is a production runtime (Soletta FBP will generate C) used by a backend process engine written in [Dao](https://github.com/daokoder/dao ). The second use case should be quite easy based on what you wrote, but the first one might be a bit tricky because of threading etc. We're also thinking about skipping the first use case in favor of super fast test deployment on a local machine which would replace the need for running it in a web browser.

In case you have some highlights, I'd be glad if you shared them with me.

barbieri, 2016-10-17:
Did you check https://github.com/solettaproject/soletta-dev-app instead of compile to JS and run "native" in the browser, we run it from a web server, calling sol-fbp-runner instead of sol-fbp-generator. Maybe it can solve the simulation part? I already thought about "compiling" soletta to JS so it run in the browser, however that would restrict the I/O and node types to be used, vastly reducing the usage... thus we remained with run on target board approach. Also, note that dev-app integrates with sol-fbp-runner webinspector (`sol-fbp-runner -W8080 file.fbp & chromium http://localhost:8080/`), that allows one to see in real-time (using HTTP Server Side Events) packets that were sent and delivered.

As for the second case, related to Dao, need more clarification as it could be used in two ways:
 1. You provide a Dao "meta-type", then one could use in the FBP a statement such as `DECLARE=MyTypeInDao:dao:script.dao`, then proceed using `MyTypeInDao` as other node types, but instead of a native code, it will call your methods, that translate to/from Dao. We do this for JavaScript using duktape (https://github.com/solettaproject/soletta/tree/master/src/modules/flow-metatype/js) This way Soletta is the *host*, Dao is used as an extension and scripting, allowing nodes to be more easily created without need of a C compiler and the likes.
 2. Use soletta's FBP node types stand-alone, or even a FBP itself from Dao. We were doing this in Node.JS and Python, but none reached that part yet -- as the groups doing those projects were more focused on using soletta's low level APIs, such as GPIO/I2C. The underlying idea is that FBP node types expose their "API" in a very simple way, by means of a constructor (`open`), destructor (`close`) and input and output ports. Thus one can use the node stand-alone, as done in my example "Single Node" (https://github.com/solettaproject/soletta/blob/master/src/samples/flow/c-api/single-node.c). You could easily expose that as a class/type in Dao, then allow instances to be created, send information (ie: a method that sends packets to input ports) and read/notify (ie: a method that delivers packets to your Dao code).

Note that both options can (should?) coexist.

The first one is useful to implement logic in FBP without having to resort to complex C build systems (ie: pkg-config, python script to read our JSON and write C boilerplate, C compiler and linker). Say you need to do a simple while/for or even more complex algorithm... you shouldn't try to do all of its internal (which are low level) in FBP using node types, these are meant for higher level logic. See our "fibonacci.fbp" example (or should I call it a "counter example"). See @glima's text on FBP and logic constructs https://github.com/solettaproject/soletta/wiki/Logic-constructs-on-Soletta-flows

And the second one exposes FBP goodies to other languages by using soletta as a guest. Say you want to use our http-server/client node types, or even gpio/reader... Of course it can use a FBP node type that was implemented another language, like JS or Dao itself :-)

dumblob, 2016-11-17:
Thank you @barbieri, this flashes some light on how to use FBP and it's high-level internals. I'm closing this issue as it answered my questions.

closed 2016-11-17 by dumblob