solettaproject/soletta #2330: iio: support oversampling_ratio configuration in iio node

laykuanloon, 2016-10-18:
@peijiajames 

[BMP280](http://lxr.free-electrons.com/source/drivers/iio/pressure/bmp280-core.c#L105) iio driver contains "oversampling_ratio", can support the configuration from iio node ? For e.g: by setting "oversampling_ration" in json file.

------------------
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_x_oversampling_ratio
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_y_oversampling_ratio
What:		/sys/bus/iio/devices/iio:deviceX/in_magn_z_oversampling_ratio
KernelVersion:	4.2
Contact:	linux-iio@vger.kernel.org
Description:
		Hardware applied number of measurements for acquiring one
		data point. The HW will do <type>[_name]_oversampling_ratio
		measurements and return the average value as output data. Each
		value resulted from <type>[_name]_oversampling_ratio measurements
		is considered as one sample for <type>[_name]_sampling_frequency.

@elvinongbl FYI

joyceooi295, 2017-01-10:
@edersondisouza @ceolin @bdilly @barbieri Please help

peijiajames, 2017-01-10:
@guchaojie is working on this already. @guchaojie please comment with the implementation.

guchaojie, 2017-01-11:
@joyceooi295  We would add this attribute into json file for user to set in iio node just like other attribute support method

yongli3, 2017-01-18:
@laykuanloon @peijiajames @guchaojie Based on kernel commit id 62979904b0037430ecc7d8b682f684adced1340f. It seems to me that only kernel 4.8+ supports this "oversampling" feature for BMP280