solettaproject/soletta #2336: rand call without initialize in sol-iio.c

yongli3, 2016-10-20:
The rand api call in sol-iio.c will generate the same random number sequence, since there is no random call

edersondisouza, 2016-10-20:
Not sure how bad is this, since it's only used to generate a different name for sysfs trigger - but patches are welcome =D
This made me think that `create_hrtimer_trigger` doesn't do any effort to create another name for hrtrigger should its rand try fail. Maybe it's worth do like on `create_sysfs_trigger` and try some times?

yongli3, 2016-10-20:
I found this issue during triggers test.  I run these trigger test .fbp files for different sensors, After several times tests, sometimes the create trigger will fail. The root cause is duplicated sysfs trigger name.
hrtimer trigger is using the same rand api to generate a random name, I think it needs change too:

INF: ./src/lib/io/sol-iio.c:406 check_trigger() No current trigger for iio:device0. Creating a sysfs one.
WRN: ./src/lib/io/sol-iio.c:328 create_sysfs_trigger() Could not create sysfs trigger.
WRN: ./src/lib/io/sol-iio.c:879 sol_iio_open() No trigger available for device0
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:324 magnet_create_channels() mdata->iio_base.device == NULL
WRN:sol-flow ./src/lib/flow/sol-flow.c:128 sol_flow_node_init() failed to create node of type=0x76c1dd28: Invalid argument
WRN:sol-flow ./src/lib/flow/sol-flow-static.c:463 flow_node_open() failed to init node #0, type=0x76c1dd28, opts=0x55357fc0: Invalid argument